SatDump/SatDump #136: NOAA HRPT Live Decoding

radio-satellites, 2022-09-21:
Hello,

First of all, I want to say that the new update is simply AWESOME!!! Honestly, I love it.

First, I was wondering about Open Street Map crashing when doing projections on windows. Also, the "add layer" button doesn't work for me, when I try to import any image (a composition), it just disappears and no layer appears. 

Second, would it be possible to support real-time HRPT decoding? It would be really cool, and a fantastic addition to an already fantastic software!

Thank you,

Sasha :)

Aang23, 2022-09-21:
Hello! Thanks :-)

What format were the images you tried to import? That should be functional.
I will take a look at the OSM bug tomorrow, probably something to do with the recent experimentations I'd say, as I noticed that on my phone (Android) yesterday.

About Live HRPT Decoding, well, that's already a thing! And in fact not for just HRPT, but most satellites that can be done on a decent PC. For live decoding, you can use the "recorder" tab. First start your SDR, then the "Live Processing" menu will let you start a live demodulator. Once done, press stop and it will finish processing as usual.

You're welcome!

I will rename this issue to rather reflect the OSM/Layer issues, as the rest should have been answered.

Aang23, 2022-09-21:
Ok. There indeed is a bug with OSM on Windows now. Will check tomorrow.

radio-satellites, 2022-09-21:
Hello! Thank you very much for confirming! :)

As for decoding, I mean like a live image on the screen. Is that already implemented? Thank you for your help!! 👍

ZbychuButItWasTaken, 2022-09-26:
Well, most decoders (all the ones I made for sure) should support a feature like that out of the box. However, utilizing that will add a lot of strain on the computer and will cause many issues on lower end machines. So for now I'd say it will not be worked on. (Definitely not on the top of the todo).

Aang23, 2022-09-26:
Yeah, can't add much to what @ZbychuButItWasTaken said.
The way it's done would allow for such features rather easily, but apart from GEO 24/7 setups, it makes a lot more sense to save frames and *then* process to imagery. It's doable, perhaps could be supported, but for how much sense it makes not entirely sure it's worth it.

Also, the bugs you had with projections should be fixed!

closed 2022-10-18 by ZbychuButItWasTaken