SatDump/SatDump #137: Error starting the gui

Botox200, 2022-09-25:
Compiled successfully, but got this message:

terminate called after throwing an instance of 'nlohmann::detail::type_error'
  what():  [json.exception.type_error.302] type must be number, but is array
Abgebrochen (Speicherabzug geschrieben)

Aang23, 2022-09-25:
Did you symlink all required files as mentioned in the README?
Eg, you should symlink/copy `satdump_cfg.json`, `pipelines` and `resources`.

Botox200, 2022-09-25:
No, i installed it systemwide with
`sudo make install`

Aang23, 2022-09-25:
Oh! Then that *is* an issue. Make install should be settings those up.

Aang23, 2022-09-25:
@Botox200 Ok, this is probably because you had the old version installed previously. The old pipelines are not compatible and will cause this sort of errors.

closed 2022-09-25 by Aang23