SatDump/SatDump #138: Raspberry Pi 4 not seeing GOES 16/17

thomasgolladay, 2022-09-26:
Howdy

For the last month, I have been trying to set up a 24/7 Satdump setup on a Raspberry Pi 4 model B but it never receives the GOES 16 and 17 signals. I have a RTL-SDR V3, a NooElec GOES SAWbird, and NooElec Goes dish. I have downloaded the new beta and is using:

"./satdump live goes_hrit /media/regs1/REGS11/GOES_output --source rtlsdr --samplerate 1792000 --frequency 1694.1e6 --general_gain 40 --bias"

as my command on my Pi. The reason I know it doesnt work is that I set up the dish with satdump on my windows computer. Then, without moving the dish, I plug in the SDR into my Pi. After running the command, I get 0 SNR. I did double check, the SAWbird does receive power from the Bias tee. I tried many different things to fix it but nothing working, including installing it many times.

Aang23, 2022-09-26:
Hmmmm... Interesting. Perhaps there's a bug I didn't notice? Let me boot up a Pi, I will try shortly (by TXing a GOES HRIT baseband locally).
I'm thinking it could be the RTL-SDR gain not getting set properly - it's a common librtlsdr issue... Perhaps try using the recorder on the Pi and check the resulting baseband?

ZbychuButItWasTaken, 2022-09-27:
Sounds like it definitely could be an issue with the sdr gain

Aang23, 2022-09-27:
@thomasgolladay I have implemented a fix for RTL-SDR Gain (it's a librtlsdr bug). Would you mind testing the latest commit?

thomasgolladay, 2022-09-27:
@Aang23 I am reinstalling satdump on my Pi right now. I had a question about this line. "# On Raspberry PIs, you will need to export MESA_GL_VERSION_OVERRIDE=4.5" What does this mean?

Aang23, 2022-09-27:
This is only if you want to use the UI. Pis apart from the Pi 4 require that override for OpenGL to start correctly in SatDump.
As you're using the CLI, you do not need that.

thomasgolladay, 2022-09-27:
@Aang23 I just ran the new code and it still didn't receive GOES 17. What command did you use for your testing with a Raspberry Pi?

ZbychuButItWasTaken, 2022-09-28:
Just to be sure: you are using *the same exact* command to test it on windows?

Aang23, 2022-09-28:
@thomasgolladay turns out I had another RTL-SDR bug. Mind testing the latest commit?

thomasgolladay, 2022-09-28:
@Aang23 It works! I used the command 

"./satdump live goes_hrit /media/regs1/REGS11/GOES_output --source rtlsdr --samplerate 1792000 --frequency 1694.1e6 --gain 40 --bias"

Of course the output file will be different for everyone but I had to use "gain 40" instead of "general_gain 40"

ZbychuButItWasTaken, 2022-09-29:
Indeed, easy to oversee. Perhaps it would be beneficial to mention that in the readme/blog @Aang23 ?

closed 2022-09-29 by ZbychuButItWasTaken

Aang23, 2022-09-29:
Well nice! So it was a gain issue all along... I don't use RTL-SDR much anymore so that's most likely why I didn't notice what wasn't as obvious.

@ZbychuButItWasTaken I have already linked to https://github.com/altillimity/SatDump/blob/master/docs/SDR-Options.md in the tutorial article.