SatDump/SatDump #142: error: ‘ZSTD_c_compressionLevel’ was not declared in this scope

K4KDR, 2022-10-06:
Hello!  Had no trouble building v1.0, but with the recent BlueWalker-3 update, I get the following on my Ubuntu 18.04 computer:

```
[ 25%] Building CXX object src-core/CMakeFiles/satdump_core.dir/core/config.cpp.o
/home/k4kdr/satdump/src-core/common/ziq.cpp: In constructor ‘ziq::ziq_writer::ziq_writer(ziq::ziq_cfg, std::ofstream&)’:
/home/k4kdr/satdump/src-core/common/ziq.cpp:27:46: error: ‘ZSTD_c_compressionLevel’ was not declared in this scope
             ZSTD_CCtx_setParameter(zstd_ctx, ZSTD_c_compressionLevel, zst_level);
```

K4KDR, 2022-10-06:
Many thanks for the reply!

For anyone who finds this in the ISSUE history, the error was cleared by installing the latest **ZSTD** from:

https://github.com/facebook/zstd/releases/tag/v1.5.2

closed 2022-10-06 by ZbychuButItWasTaken

ZbychuButItWasTaken, 2022-10-06:
Closed for now as seems it was just outdated zstd, could be reopened if needed