SatDump/SatDump #144: suggestion - provide the option to store CADU files when use live processing

haiyangdaozhang, 2022-10-10:
Hello,

I found that the program saves the CADU data only when the baseband data is stored and then decoded offline. Is there an option to save CADU files during live decoding?

Thanks for this software!

ZbychuButItWasTaken, 2022-10-10:
Which pipeline are you using? CADU should be saved

haiyangdaozhang, 2022-10-10:
fengyun svissr and gk2a lrit

ZbychuButItWasTaken, 2022-10-10:
Ah yes, we do not save the cadu on these as they are meant to be run long term (which would make the cadu massive). Howevee we can think of an option to save it @Aang23

Aang23, 2022-10-11:
Indeed, currently pipelines that process things down to images directly do not save the CADU. Would you want to "simply" disable said final image processing or rather dump the cadu alongside decoding images?

haiyangdaozhang, 2022-10-12:
My suggestion is to add a button that allows dump the cadu alongside decoding images.