SatDump/SatDump #146: Remote use of Satdump

hanslammerts, 2022-10-11:
First of all,
Thank you for your hard work ! Compilation on Debian 11 was straightforward.

My setup:
I have a linux machine that has all the RTL-SDRs connected to, and my desktop Linux machine is another Debian 11 machine that runs Satdump.
I would like to use SatDump with the remote RTL-SDR.
Have tried rtl_tcp and sdr-server. Both work when I connect using GQRX, but I can't seem to figure out if the same is possible with SatDump.

Any suggestions, please ?

Aang23, 2022-10-12:
Hi!

Well, looks like I forgot to add back rtl_tcp support... Will do soon enough.

Aang23, 2022-10-19:
As a note, I've just added SDR++ Server support which could also help.

Aang23, 2022-10-24:
Added back RTL-TCP.

closed 2022-10-24 by Aang23