SatDump/SatDump #147: parts of image missing in 221 RGB composite

Ajaykumar415, 2022-10-12:
So I've been trying to use satdump v1.0.0 and every time that  I get a pass and make a composite image a lot of the image goes missing but this did not seem to happen with v0.0.3B
[image]
[image]

Aang23, 2022-10-12:
Hm. That's weird behaviour indeed. Do you have the files on hand? (frames / soft / baseband)

Ajaykumar415, 2022-10-12:
No as it was a live decoding session.

Aang23, 2022-10-12:
Live decoding should keep the .cadu file though?

Ajaykumar415, 2022-10-13:
yup i do have the cadu file but can't upload it here, it seems git hub doesn't support this file format.

RAD750, 2022-10-13:
It would look to me like you have some sort of processing power issue, watch cpu usage and if on Linux, make sure to compile SatDump properly or it will run very slow

Ajaykumar415, 2022-10-14:
I am on windows, and I have set satdump to use CUDA and my computer does have a GTX1650Ti, i will check the GPU and CPU usage in the next pass. I will look into uploading the CADU file.

ZbychuButItWasTaken, 2022-10-21:
Any update on the issue?

Aang23, 2022-11-24:
Well, since there were no updates I will close this. Will re-open if necessary.

closed 2022-11-24 by Aang23