SatDump/SatDump #151: Suggestion: Add image processing options to CLI utility

sys-256, 2022-10-15:
Hi, thanks a lot for satdump v1.0.0, it works great!

Now, I want to make a bash script that creates 221 (or 321) images from LRPT .s files. I can run `satdump meteor_m2_lrpt soft ./file.s output` to get the .cadu file and individual channels, but creating white balanced and/or equalized images isn't possible with v1.0.0, while it is with v0.0.38.

Would it be possible to expose the "viewer" tab as an extra utility (or integrated within the current satdump command) in a future release?

Aang23, 2022-10-16:
Actually, that should already be there, if you enable products processing.

sys-256, 2022-10-17:
Hmm, okay. I set `satdump_general->auto_process_products->value` in `~/.config/satdump/settings.json` to `true`, but since it's the default too it didn't make any difference. Only the `.cadu` file and individual channels were created, not the white balanced and/or equalized RGB 221 images I hoped for, nor any inverted ch5 ones.

I used the latest commit btw, and built it from source with the `satdump-git` AUR package.

Here's the output:
```
╭─ /home/bram/SDR/weather-imaging/Meteor-M_2/15-10_copy 
╰─ $ ls
meteor_08-43-35_15-10-2022.s

╭─ /home/bram/SDR/weather-imaging/Meteor-M_2/15-10_copy 
╰─ $ cat ~/.config/satdump/settings.json
{
    "satdump_general": {
        "update_tles_startup": {
            "value": true
        },
        "auto_process_products": {
            "value": true
        }
    },
    "user": {
        "recorder_state": {
            "fft_avg": 0.009999999776482582,
            "fft_max": 20.0,
            "fft_min": -10.0,
            "panel_ratio": 0.20000000298023224,
            "show_waterfall": true,
            "waterfall_ratio": 0.699999988079071
        }
    }
}

╭─ /home/bram/SDR/weather-imaging/Meteor-M_2/15-10_copy 
╰─ $ satdump meteor_m2_lrpt soft ./meteor_08-43-35_15-10-2022.s satdump
[10/17/22 - 11:02:02] (I)    _____       __  ____                      
[10/17/22 - 11:02:02] (I)   / ___/____ _/ /_/ __ \__  ______ ___  ____ 
[10/17/22 - 11:02:02] (I)   \__ \/ __ `/ __/ / / / / / / __ `__ \/ __ \
[10/17/22 - 11:02:02] (I)  ___/ / /_/ / /_/ /_/ / /_/ / / / / / / /_/ /
[10/17/22 - 11:02:02] (I) /____/\__,_/\__/_____/\__,_/_/ /_/ /_/ .___/ 
[10/17/22 - 11:02:02] (I)                                     /_/      
[10/17/22 - 11:02:02] (I) Starting SatDump v1.0.0
[10/17/22 - 11:02:02] (I) 
[10/17/22 - 11:02:02] (I) Loading config /usr/share/satdump//satdump_cfg.json
[10/17/22 - 11:02:02] (I) Loading user settings /home/bram/.config/satdump/settings.json
[10/17/22 - 11:02:02] (I) Loading plugins from /usr/share/satdump//plugins
[10/17/22 - 11:02:02] (T) Loading plugin /usr/share/satdump//plugins/libeos_support.so...
[10/17/22 - 11:02:02] (T) Plugin eos_support loaded!
[10/17/22 - 11:02:02] (T) Loading plugin /usr/share/satdump//plugins/libelektro_arktika_support.so...
[10/17/22 - 11:02:02] (T) Plugin elektro_arktika_support loaded!
[10/17/22 - 11:02:02] (T) Loading plugin /usr/share/satdump//plugins/libplutosdr_sdr_support.so...
[10/17/22 - 11:02:02] (T) Plugin plutosdr_sdr_support loaded!
[10/17/22 - 11:02:02] (T) Loading plugin /usr/share/satdump//plugins/libhimawari_support.so...
[10/17/22 - 11:02:02] (T) Plugin himawari_support loaded!
[10/17/22 - 11:02:02] (T) Loading plugin /usr/share/satdump//plugins/libothers_support.so...
[10/17/22 - 11:02:02] (T) Plugin others_support loaded!
[10/17/22 - 11:02:02] (T) Loading plugin /usr/share/satdump//plugins/libnoaa_metop_support.so...
[10/17/22 - 11:02:02] (T) Plugin noaa_metop_support loaded!
[10/17/22 - 11:02:02] (T) Loading plugin /usr/share/satdump//plugins/libjpss_support.so...
[10/17/22 - 11:02:02] (T) Plugin jpss_support loaded!
[10/17/22 - 11:02:02] (T) Loading plugin /usr/share/satdump//plugins/libbladerf_sdr_support.so...
[10/17/22 - 11:02:02] (T) Plugin bladerf_sdr_support loaded!
[10/17/22 - 11:02:02] (T) Loading plugin /usr/share/satdump//plugins/libhackrf_sdr_support.so...
[10/17/22 - 11:02:02] (T) Plugin hackrf_sdr_support loaded!
[10/17/22 - 11:02:02] (T) Loading plugin /usr/share/satdump//plugins/libmirisdr_sdr_support.so...
[10/17/22 - 11:02:02] (T) Plugin mirisdr_sdr_support loaded!
[10/17/22 - 11:02:02] (T) Loading plugin /usr/share/satdump//plugins/libspyserver_support.so...
[10/17/22 - 11:02:02] (T) Plugin spyserver_support loaded!
[10/17/22 - 11:02:02] (T) Loading plugin /usr/share/satdump//plugins/libgk2a_support.so...
[10/17/22 - 11:02:02] (T) Plugin gk2a_support loaded!
[... 316 lines cut ...]
[10/17/22 - 11:02:42] (I) MSU-MR Channel 3 Lines  : 0
[10/17/22 - 11:02:42] (I) MSU-MR Channel 4 Lines  : 0
[10/17/22 - 11:02:42] (I) MSU-MR Channel 5 Lines  : 2200
[10/17/22 - 11:02:42] (I) MSU-MR Channel 6 Lines  : 0
[10/17/22 - 11:02:42] (I) Writing images.... (Can take a while)
[10/17/22 - 11:02:42] (I) Saving MSU-MR-1.png
[10/17/22 - 11:02:42] (I) Saving MSU-MR-2.png
[10/17/22 - 11:02:42] (I) Saving MSU-MR-5.png
[10/17/22 - 11:02:43] (I) Done! Goodbye

╭─ /home/bram/SDR/weather-imaging/Meteor-M_2/15-10_copy 
╰─ $ tree .
.
├── meteor_08-43-35_15-10-2022.s
├── satdump
│   ├── meteor_m2_lrpt.cadu
│   └── MSU-MR
│       ├── MSU-MR-1.png
│       ├── MSU-MR-2.png
│       ├── MSU-MR-5.png
│       └── product.cbor
└── satdump.logs

2 directories, 7 files
````

Aang23, 2022-10-17:
Oh crap. I see why.

Somehow forgot to make LRPT write a dataset file. Said file is what allows the next processing step to be called and create composites etc.

sys-256, 2022-10-17:
Ah okay. Thanks for the help!

Aang23, 2022-10-19:
@sys-256 should be fixed now.

sys-256, 2022-10-19:
It partially worked, as it crashes when trying to write a file with the 3rd channel, but that band is currently deactivated to make place for band 5 (thermal infrared).

```
[10/19/22 - 20:29:30] (I) Starting processing pipeline meteor_m2_lrpt...
[10/19/22 - 20:29:30] (D) Input file (soft) : ./meteor_08-43-35_15-10-2022.s
[10/19/22 - 20:29:30] (D) Output file : satdump
[10/19/22 - 20:29:30] (D) Starting meteor_m2_lrpt
[10/19/22 - 20:29:30] (W) Data is already at level baseband, skipping
[10/19/22 - 20:29:30] (W) Data is already at level soft, skipping
[10/19/22 - 20:29:30] (W) Processing data to level cadu
[10/19/22 - 20:29:30] (D) Parameters :
[10/19/22 - 20:29:30] (D)    - diff_decode : false
[10/19/22 - 20:29:30] (T) Volk has the spiral kernel, using it!
[10/19/22 - 20:29:30] (I) Using input symbols ./meteor_08-43-35_15-10-2022.s
[10/19/22 - 20:29:30] (I) Decoding to satdump/meteor_m2_lrpt.cadu
[10/19/22 - 20:29:30] (I) Progress 0.000000%, Viterbi BER : 25.585938%, Lock : NOSYNC
[10/19/22 - 20:29:40] (I) Progress 15.100000%, Viterbi BER : 26.562500%, Lock : NOSYNC
[10/19/22 - 20:29:50] (I) Progress 31.400000%, Viterbi BER : 24.414062%, Lock : NOSYNC
[10/19/22 - 20:30:00] (I) Progress 89.699997%, Viterbi BER : 26.757812%, Lock : NOSYNC
[10/19/22 - 20:30:06] (W) Processing data to level products
[10/19/22 - 20:30:06] (D) Parameters :
[10/19/22 - 20:30:06] (I) Using input frames satdump/meteor_m2_lrpt.cadu
[10/19/22 - 20:30:06] (I) Decoding to satdump/MSU-MR
[10/19/22 - 20:30:07] (I) Demultiplexing and deframing...
[10/19/22 - 20:30:09] (I) MSU-MR Channel 1 Lines  : 2208
[10/19/22 - 20:30:09] (I) MSU-MR Channel 2 Lines  : 2208
[10/19/22 - 20:30:09] (I) MSU-MR Channel 3 Lines  : 0
[10/19/22 - 20:30:09] (I) MSU-MR Channel 4 Lines  : 0
[10/19/22 - 20:30:09] (I) MSU-MR Channel 5 Lines  : 2200
[10/19/22 - 20:30:09] (I) MSU-MR Channel 6 Lines  : 0
[10/19/22 - 20:30:09] (I) Writing images.... (Can take a while)
[10/19/22 - 20:30:09] (I) Saving MSU-MR-1.png
[10/19/22 - 20:30:10] (I) Saving MSU-MR-2.png
[10/19/22 - 20:30:11] (I) Saving MSU-MR-5.png
[10/19/22 - 20:30:11] (D) Products processing is enabled! Starting processing module.
[10/19/22 - 20:30:11] (I) satdump/MSU-MR
[10/19/22 - 20:30:11] (I) satdump/MSU-MR/product.cbor
[10/19/22 - 20:30:11] (I) satdump/MSU-MR/product.cbor
[10/19/22 - 20:30:11] (I) Loading MSU-MR-1.png
[10/19/22 - 20:30:11] (I) Loading MSU-MR-2.png
[10/19/22 - 20:30:11] (I) Loading MSU-MR-5.png
[10/19/22 - 20:30:11] (D) Composite needs channel ch1
[10/19/22 - 20:30:11] (D) Composite needs channel ch2
[10/19/22 - 20:30:11] (T) Max Total Width 1568
[10/19/22 - 20:30:11] (T) Max Total Used  1568
[10/19/22 - 20:30:11] (T) Offset for ch1 is 0
[10/19/22 - 20:30:11] (T) Offset for ch2 is 0
[10/19/22 - 20:30:12] (I) Saving satdump/MSU-MR/msu_mr_rgb_221.png
[10/19/22 - 20:30:13] (D) Composite needs channel ch1
[10/19/22 - 20:30:13] (D) Composite needs channel ch2
[10/19/22 - 20:30:13] (T) Max Total Width 1568
[10/19/22 - 20:30:13] (T) Max Total Used  1568
[10/19/22 - 20:30:13] (T) Offset for ch1 is 0
[10/19/22 - 20:30:13] (T) Offset for ch2 is 0
[10/19/22 - 20:30:13] (E) Unexpected token "ch3" found at position 0.
[10/19/22 - 20:30:13] (I) Saving satdump/MSU-MR/msu_mr_rgb_321.png
[10/19/22 - 20:30:13] (T) Tried to save empty PNG!
[10/19/22 - 20:30:13] (D) Reprojecting composite msu_mr_rgb_321
[10/19/22 - 20:30:14] (I) Using old algorithm...
[10/19/22 - 20:30:14] (E) Fatal error running pipeline : [json.exception.type_error.302] type must be string, but is null

╭─ /home/bram/SDR/weather-imaging/Meteor-M_2/15-10_copy 
╰─ $ tree .
.
├── meteor_08-43-35_15-10-2022.s
├── satdump
│   ├── dataset.json
│   ├── meteor_m2_lrpt.cadu
│   └── MSU-MR
│       ├── MSU-MR-1.png
│       ├── MSU-MR-2.png
│       ├── MSU-MR-5.png
│       ├── msu_mr_rgb_221.png
│       └── product.cbor
└── satdump.logs

2 directories, 9 files

╭─ /home/bram/SDR/weather-imaging/Meteor-M_2/15-10_copy 
╰─ $ cat satdump/dataset.json
{
    "products": [
        "MSU-MR"
    ],
    "satellite": "METEOR-M 2",
    "timestamp": 1666204209.0
}
```

Aang23, 2022-10-19:
Ah oops. That's not a crash with that, just another oversight. Sorry!
Gonna fix in a bit.

Aang23, 2022-10-21:
Fixed!

closed 2022-10-21 by Aang23

sys-256, 2022-10-21:
It worked for me too. Thanks a lot!

Aang23, 2022-10-21:
You're welcome!