SatDump/SatDump #155: Cannot decompress received images. "Rice decompression failed. This may be an issue!"

cosmopotamian, 2022-10-19:
I am running Windows 11 and it works pretty well on it! Until a point where it starts decompressing image data. It crashes and quits. I tried everything, from compatibility with order windows versions, to running it as admin, to granting full write access to everyone for the main satdump folder and its subfolders.. 

Here is the log:

[10/19/22 - 17:45:50] (E) Usage : satdump-ui.exe [downlink] [input_level] [input_file] [output_file_or_directory] [additional options as required]
[10/19/22 - 17:45:50] (E) Extra options (examples. Any parameter used in modules can be used here) :
[10/19/22 - 17:45:50] (E)  --samplerate [baseband_samplerate] --baseband_format [f32/i16/i8/w8] --dc_block --iq_swap
[10/19/22 - 17:45:50] (I)    _____       __  ____                      
[10/19/22 - 17:45:50] (I)   / ___/____ _/ /_/ __ \__  ______ ___  ____ 
[10/19/22 - 17:45:50] (I)   \__ \/ __ `/ __/ / / / / / / __ `__ \/ __ \
[10/19/22 - 17:45:50] (I)  ___/ / /_/ / /_/ /_/ / /_/ / / / / / / /_/ /
[10/19/22 - 17:45:50] (I) /____/\__,_/\__/_____/\__,_/_/ /_/ /_/ .___/ 
[10/19/22 - 17:45:50] (I)                                     /_/      
[10/19/22 - 17:45:50] (I) Starting SatDump v1.0.0
[10/19/22 - 17:45:50] (I) 
[10/19/22 - 17:45:50] (I) Loading config satdump_cfg.json
[10/19/22 - 17:45:50] (I) Loading user settings settings.json
[10/19/22 - 17:45:50] (I) Loading plugins from ./plugins
[10/19/22 - 17:45:50] (T) Loading plugin ./plugins\airspyhf_sdr_support.dll...
[10/19/22 - 17:45:50] (T) Plugin airspyhf_sdr_support loaded!
[10/19/22 - 17:45:50] (T) Loading plugin ./plugins\airspy_sdr_support.dll...
[10/19/22 - 17:45:50] (T) Plugin airspy_sdr_support loaded!
[10/19/22 - 17:45:50] (T) Loading plugin ./plugins\bladerf_sdr_support.dll...
[10/19/22 - 17:45:50] (T) Plugin bladerf_sdr_support loaded!
[10/19/22 - 17:45:50] (T) Loading plugin ./plugins\elektro_arktika_support.dll...
[10/19/22 - 17:45:50] (T) Plugin elektro_arktika_support loaded!
[10/19/22 - 17:45:50] (T) Loading plugin ./plugins\eos_support.dll...
[10/19/22 - 17:45:50] (T) Plugin eos_support loaded!
[10/19/22 - 17:45:50] (T) Loading plugin ./plugins\fengyun2_support.dll...
[10/19/22 - 17:45:50] (T) Plugin fengyun2_support loaded!
[10/19/22 - 17:45:50] (T) Loading plugin ./plugins\fengyun3_support.dll...
[10/19/22 - 17:45:50] (T) Plugin fengyun3_support loaded!
[10/19/22 - 17:45:50] (T) Loading plugin ./plugins\gk2a_support.dll...
[10/19/22 - 17:45:50] (T) Plugin gk2a_support loaded!
[10/19/22 - 17:45:50] (T) Loading plugin ./plugins\goes_support.dll...
[10/19/22 - 17:45:50] (T) Plugin goes_support loaded!
[10/19/22 - 17:45:50] (T) Loading plugin ./plugins\hackrf_sdr_support.dll...
[10/19/22 - 17:45:50] (T) Plugin hackrf_sdr_support loaded!
[10/19/22 - 17:45:50] (T) Loading plugin ./plugins\himawari_support.dll...
[10/19/22 - 17:45:50] (T) Plugin himawari_support loaded!
[10/19/22 - 17:45:50] (T) Loading plugin ./plugins\jason3_support.dll...
[10/19/22 - 17:45:50] (T) Plugin jason3_support loaded!
[10/19/22 - 17:45:50] (T) Loading plugin ./plugins\jpss_support.dll...
[10/19/22 - 17:45:50] (T) Plugin jpss_support loaded!
[10/19/22 - 17:45:50] (T) Loading plugin ./plugins\limesdr_sdr_support.dll...
[10/19/22 - 17:45:50] (T) Plugin limesdr_sdr_support loaded!
[10/19/22 - 17:45:50] (T) Loading plugin ./plugins\meteor_support.dll...
[10/19/22 - 17:45:50] (T) Plugin meteor_support loaded!
[10/19/22 - 17:45:50] (T) Loading plugin ./plugins\mirisdr_sdr_support.dll...
[10/19/22 - 17:45:50] (T) Plugin mirisdr_sdr_support loaded!
[10/19/22 - 17:45:50] (T) Loading plugin ./plugins\noaa_metop_support.dll...
[10/19/22 - 17:45:50] (T) Plugin noaa_metop_support loaded!
[10/19/22 - 17:45:50] (T) Loading plugin ./plugins\oceansat_support.dll...
[10/19/22 - 17:45:50] (T) Plugin oceansat_support loaded!
[10/19/22 - 17:45:50] (T) Loading plugin ./plugins\others_support.dll...
[10/19/22 - 17:45:50] (T) Plugin others_support loaded!
[10/19/22 - 17:45:50] (T) Loading plugin ./plugins\plutosdr_sdr_support.dll...
[10/19/22 - 17:45:50] (T) Plugin plutosdr_sdr_support loaded!
[10/19/22 - 17:45:50] (T) Loading plugin ./plugins\proba_support.dll...
[10/19/22 - 17:45:50] (T) Plugin proba_support loaded!
[10/19/22 - 17:45:50] (T) Loading plugin ./plugins\rtlsdr_sdr_support.dll...
[10/19/22 - 17:45:50] (T) Plugin rtlsdr_sdr_support loaded!
[10/19/22 - 17:45:50] (T) Loading plugin ./plugins\sdrplay_sdr_support.dll...
[10/19/22 - 17:45:50] (E) Could not open the SDRPlay API, perhaps the service is not running?
[10/19/22 - 17:45:50] (T) Plugin sdrplay_sdr_support loaded!
[10/19/22 - 17:45:50] (T) Loading plugin ./plugins\spacex_support.dll...
[10/19/22 - 17:45:50] (T) Plugin spacex_support loaded!
[10/19/22 - 17:45:50] (T) Loading plugin ./plugins\spyserver_support.dll...
[10/19/22 - 17:45:50] (T) Plugin spyserver_support loaded!
[10/19/22 - 17:45:50] (D) Loaded plugins (25) : 
[... 560 lines cut ...]
[10/19/22 - 17:47:07] (W) Rice decompression failed. This may be an issue!
[10/19/22 - 17:47:07] (W) Rice decompression failed. This may be an issue!
[10/19/22 - 17:47:07] (W) Rice decompression failed. This may be an issue!
[10/19/22 - 17:47:07] (W) Rice decompression failed. This may be an issue!
[10/19/22 - 17:47:07] (W) Rice decompression failed. This may be an issue!
[10/19/22 - 17:47:07] (W) Rice decompression failed. This may be an issue!
[10/19/22 - 17:47:07] (W) Rice decompression failed. This may be an issue!
[10/19/22 - 17:47:08] (W) Rice decompression failed. This may be an issue!
[10/19/22 - 17:47:08] (W) Rice decompression failed. This may be an issue!
[10/19/22 - 17:47:08] (W) Rice decompression failed. This may be an issue!
[10/19/22 - 17:47:08] (W) Rice decompression failed. This may be an issue!
[10/19/22 - 17:47:08] (W) Rice decompression failed. This may be an issue!
[10/19/22 - 17:47:08] (W) Rice decompression failed. This may be an issue!
[10/19/22 - 17:47:08] (W) Rice decompression failed. This may be an issue!
[10/19/22 - 17:47:08] (W) Rice decompression failed. This may be an issue!
[10/19/22 - 17:47:08] (W) Rice decompression failed. This may be an issue!
[10/19/22 - 17:47:08] (W) Rice decompression failed. This may be an issue!
[10/19/22 - 17:47:08] (W) Rice decompression failed. This may be an issue!
[10/19/22 - 17:47:08] (W) Rice decompression failed. This may be an issue!
[10/19/22 - 17:47:08] (W) Rice decompression failed. This may be an issue!
[10/19/22 - 17:47:08] (W) Rice decompression failed. This may be an issue!
[10/19/22 - 17:47:08] (W) Rice decompression failed. This may be an issue!
[10/19/22 - 17:47:08] (W) Rice decompression failed. This may be an issue!
[10/19/22 - 17:47:08] (W) Rice decompression failed. This may be an issue!
[10/19/22 - 17:4

Aang23, 2022-10-19:
Interesting. Could you try another samplerate on the RTL-SDR maybe? Do you also have a screenshot of SatDump running? I'd like to see a few things.

cosmopotamian, 2022-10-19:
> 

Forgot to mention, running v1.0.0 under Linux works pretty well, after compiling my own version. It is only the Windows version that crashes.

I certainly tried using different sampling rates (down to 2,048MSPS on RTL-SDR). Same issue. Too low sample rates would go below the required so I would not be able to get a lock anyway.
I also tried recording some 3GB f32 file and tried offline-decoding that later. It still crashes at the decoding stage.

Attached a screenshot. My SNR maxes at ~8dB and I am decoding GOES-R - HRIT. Signal is certainly good enough and it decodes pretty nicely on Linux.

[image]

Aang23, 2022-10-19:
Ok, thanks. Just wanted to rule out the obvious / common user mistakes...

As long as RS is not red, it should decode anyway. So that is ~6db SNR more than required to get data out... So yeah, defintiely not the issue. I'm suspecting something related to Windows.
Could you send over a CADU (such as the one from said 3GB baseband) from Windows? I will check it on my end, and if that does not work, boot a Windows machine and test and my end. Got other Windows bugs to deal with anyway.

Definitely a bug. My thought is : either there's another sneaky bug in libaec on Windows, that'd be the third, or I forgot to manually initialize some variable somewhere leading to invalid decompression parameters.

cosmopotamian, 2022-10-19:
> Ok, thanks. Just wanted to rule out the obvious / common user mistakes...
>
> As long as RS is not red, it should decode anyway. So that is ~6db SNR more than required to get data out... So yeah, [...]

Of course! Here is a 2.5GB f32 dumped file. 
https://drive.google.com/open?id=13Bmv7tdTYesnrZRA_4aSjMVMo1wpTqCl 

BTW, I have tested offline decoding the above with another windows machine (virtualized this time), and it fails again. Seems to be consistent and has something to do with the decoding stage itself.

Aang23, 2022-10-19:
Thanks!
It's a bit weird, as I also know a few people using it sucessfully on Windows machines. Perhaps happens in a more specific case or in newer builds? One way to find out.

Aang23, 2022-10-19:
Ah, I can't download from your drive link? Only invites me to "Sign in to drop in files to this folder".

cosmopotamian, 2022-10-19:
> https://drive.google.com/open?id=13Bmv7tdTYesnrZRA_4aSjMVMo1wpTqCl

Please try it again. I tried it just now in incognito and could see the file.

Aang23, 2022-10-20:
Ok! Definitely can't be HW-related to me... Or perhaps in a weird way, but then that is a bug anyway.

What is the HW though, just in case? Not the issue but may as well be sure... Even on old HW should work just fine, even pre-Core-2-Duo era CPUs should do fine.

Aang23, 2022-10-21:
Hm well, I can't reproduce in my W10 VM, nor on a laptop I installed Windows on to test another bug...

I guess I'll test on a few more machines to see!

ZbychuButItWasTaken, 2022-10-21:
I could test on my machine too @Aang23!

Aang23, 2022-10-21:
@ZbychuButItWasTaken good idea!

Aang23, 2022-10-26:
@cosmopotamian this may have been fixed in the commit I just pushed.

cosmopotamian, 2022-10-26:
> @cosmopotamian this may have been fixed in the commit I just pushed.

Thank you for the heads up! I will test it out and let you know as soon as I am done!

Regarding your question on the specifications, the system where the testing is failing is an old Asus G75VW laptop (Republic of Gamers) with Intel Core i7 3610QM and Geforce GTX 670M and 24GB of RAM. The new system where offline decoding worked packs an Intel Core i7 4790K, 16GB of RAM and a Geforce GTX 1650. Both systems run Windows 11.

I hope that helps though it sounds like I am late to the party! :)

Aang23, 2022-10-26:
Ok... That still makes no sense.

I tried it on Windows 11 too, no issues. Could you check perhaps the specific builds both systems are running? I could try that, in case it's potentially the issue.

creas002, 2022-11-04:
I'm having the same issue. Satdump is crashing when trying to decode images with Rice Compression. I tried offline decoding using the the recorded file [cosmopotamian](https://github.com/cosmopotamian) attached in google docs and it crashes for me as well.  21-56-12_2400000SPS_1694100000Hz.f32

I'm attaching one of my log files that I copied/pasted together as the OS buffer never dumps the remainder to the log once satdump crashes.

Log from LIVE processing which I annotated using "(comments)" and copy/paste from command window when UI was launched.
[satdump1.log](https://github.com/altillimity/SatDump/files/9940946/satdump1.log)

Log from OFFLINE processing of 21-56-12_2400000SPS_1694100000Hz.f32
[satdump (3).log](https://github.com/altillimity/SatDump/files/9940963/satdump.3.log)

Aang23, 2022-11-04:
Oh nice, a second report :-)
@creas002, what hardware / OS are you running, and exact build of SatDump?

creas002, 2022-11-04:
> Oh nice, a second report :-) @creas002, what hardware / OS are you running, and exact build of SatDump?

Satdump version v1.0.0
Windows 10 Version 21H1 (OS Build 19043.1706) - Version 10.0.19043 Build 19043

Intel i7 920 @ 2..67GHz
ASUS P6T Deluxe Rev 1 - BIOS Version/Date American Megatrends Inc. 2209, 9/21/2010
Hardware Abstraction Layer	Version = "10.0.19041.1566"
Installed Physical Memory (RAM) 12.0 GB
Total Physical Memory	12.0 GB
Available Physical Memory	4.97 GB
Total Virtual Memory	20.0 GB
Available Virtual Memory	6.14 GB

creas002, 2022-11-04:
[Aang23](https://github.com/Aang23)
  Looking at Windows Event Viewer for the last time Satdump crashed I gleamed this little nugget

aulting application name: satdump-ui.exe, version: 0.0.0.0, time stamp: 0x635ab7cb
Faulting module name: VCRUNTIME140.dll, version: 14.32.31332.0, time stamp: 0xefff39ad
Exception code: 0xc0000005
Fault offset: 0x000000000000174a
Faulting process id: 0x364c
Faulting application start time: 0x01d8f082c61fbd91
Faulting application path: D:\zips\satdump\satdump-ui.exe
Faulting module path: C:\WINDOWS\SYSTEM32\VCRUNTIME140.dll
Report Id: a2d41689-d5f9-4f96-bc87-891c571c95d2
Faulting package full name: 
Faulting package-relative application ID: 

You can see my Visual C++ version in the attach picture.
[image]

creas002, 2022-11-04:
[Aang23](https://github.com/Aang23)

Not sure how much it helps, but for the first time during a LIVE capture it actually started receiving and displaying a Rice Compressed image. It may have displayed ~5% of the image before crashing. No image file was actually written to disk before it crashed. Not even an empty filename

[Aang23](https://github.com/Aang23)

Is there any code profiling support that can be enabled or a DEBUG mode that I can use to capture what exactly was being executed when it goes sideways?

Thx

creas002, 2022-11-06:
I tried SatDump again from a different Windows 10 PC and it successfully handled Rice Compressed Images without crashing, therefore it's something specific to my other PC. Off topic, it would hang during startup and could not be killed without rebooting. The cause of the hang was because it found SDRplayAPI when starting and assumed I was going to use it. The service was running, but the SDRPlay SDR was turned off. I disabled the SDRplayAPI service and rebooted and it still tried using it and just hung. I the removed the SDRplayAPI service and rebooted and SatDump ran as expected.

Aang23, 2022-11-24:
Ok @creas002 @cosmopotamian... I've tried to reproduce this on pretty much every Windows machine I could, to no avail. Without having access to a machine exhibiting the issue, it's extremely complicated to debug.

Might be asking a bit much but, is there any chance someone with the issue currently happening could spend a bit more time helping me to figure it out? (Perhaps joining the matrix room in the readme so it's a bit easier?)
Otherwise honestly, not sure what else I can do since it's been so complicated to reproduce and I can't see anything wrong when I check locally.

creas002, 2022-11-24:
I’m sorry you spent anymore time on it. I had a follow up comment that the problem did not happen on another windows machine, therefore concluded it was unquie to my PC.

Thanks for looking, but I suspect it was incompatibilities with my visual studio installation and libraries.

Regards
Richard

JVital2013, 2022-11-30:
@Aang23 I'm not having this issue, but I'm getting questions from folks that are. One thing I noticed is that everyone with the issue so far is using an Intel processor that is 3rd gen or older. There are confirmed reports of it working on 4th gen (Haswell) or newer.

It looks like your code relies heavily on AVX2 in many situations - and AVX2 was only introduced with Haswell. Are the Windows builds relying on AVX2?

creas002, 2022-11-30:
Hi Jamie,
  I will do my best to help. I'm new to SatDump, so you may need a bit of
patience. I have goesbetween running on my GOES16 RPi and Satdump running
on Windows 10 box.

I'm still getting Rice Compression errors, but its not crashing now. LRIT
files are captured fine, but images start to display and then stop.

If there is debug or logging that needs enabled, just tell me what to do
and where to find the logs.

Thanks
Richard - AI4Y
[image: hardware_os_details.jpg]

creas002, 2022-11-30:
Update - It attempted to capture multiple images and eventually crashed.

Aang23, 2022-12-01:
@JVital2013 it is reliant, but will automatically detect AVX2 is not present and not use it. But perhaps something of the sort is not the case about libaec, used for decompressing GOES imagery. I've also sometimes seen some Windows API calls breaking depending on AVX / etc support.

Should be easy to fix once I am able to reproduce.

Thanks for narrowing the issue down to older CPUs. Now that you mention it - seems obvious, and that's one thing I have not tried yet. I will install Windows on a 2nd gen I have around tomorrow and see!

JVital2013, 2022-12-02:
I scrounged up a laptop with an Arrandale CPU (Core i5 M450), and sure enough I get the problem as soon as I receive RICE-compressed data. 

It works fine until then, other than the fact that I need to run SatDump in CLI mode since I have no OpenGL support with Windows 10 on this chip. 

If you want remote access to this machine, let me know how we can securely get in touch so I can arrange a connection for you.

Aang23, 2022-12-02:
Oh, great. So now there is a reliable way to reproduce the issue. You were faster than me in the end.

About OpenGL, you can drop in this DLL (64-bits) next to SatDump and it will work. https://fdossena.com/?p=mesa/index.frag

I guess at this stage a remote connexion will be faster than me getting a machine setup from scratch.
I guess you can either join the Matrix room in the README and DM me there, otherwise there is TW or aang23@altillimity.com.

Thanks for the help figuring this one though! It was especially tough to make the link to older CPUs without a larger set of reports. Glad there's finally something to work off.

JVital2013, 2022-12-02:
@Aang23 sent an email

Aang23, 2022-12-03:
It appears I was able to fix this issue. Tested on @JVital2013's machine... So hopefully this works for everyone else affected. I will only close once someone else is able to confirm just in case.

Thanks @JVital2013 for letting me debug remotely on his laptop :-)

creas002, 2022-12-03:
I'm guessing its a bit more is required than just dropping opengl32.dll into the satdump directory 

[image]

JVital2013, 2022-12-03:
The software OpenGL driver is a separate fix for a different issue I ran into. In your case, I don't think you should use the software OpenGL driver.

To test the RICE bug fix, pull the latest build from the Github CI Builds. Here's the latest link (must be signed into GitHub): [https://github.com/altillimity/SatDump/suites/9642468624/artifacts/460957153](https://github.com/altillimity/SatDump/suites/9642468624/artifacts/460957153)

To find these builds in the future, go to Actions (at the top of this GitHub repository) > Windows Builds (On the left). Click on the top-most commit, then scroll to the bottom of the page to download the "Artifact"

[image]

creas002, 2022-12-03:
Thank you for the clarification and where to find the fix.

creas002, 2022-12-03:
I can confirm that your fix  ALMOST worked for me. It was almost at the end of capturing the first image when it crashed. I renamed the satdump.logs to a TXT file so I could attach it. It crashed so the error was not written to the log by the OS. Is there a debug mode I can run that would dump the error perhaps?

[satdump.txt](https://github.com/altillimity/SatDump/files/10147337/satdump.txt)

Aang23, 2022-12-05:
@creas002 can you open another issue about this? Can't see much in the logs and it's definitely unrelated.

closed 2022-12-07 by Aang23