SatDump/SatDump #156: Tuner gain level does not apply at the start. Need to move the slider back and forth for signal to go up!

cosmopotamian, 2022-10-19:
I am using RTL-SDR and noticed that when I set the gain to max (49.6dB and without checking AGC) and hit start, the signal starts weak. I need to move the slider back and forth to get the gain set properly for me to receive the expected signal.

Aang23, 2022-10-19:
What version are you using? I have fixed a very, very similar RTL-SDR bug a while ago.

cosmopotamian, 2022-10-19:
> What version are you using? I have fixed a very, very similar RTL-SDR bug a while ago.

It's the latest beta (v1.0.0)

Aang23, 2022-10-19:
Can you try using this build? The latest beta is a bit outdated by this point, quite some things have been addressed in the meantime.
https://github.com/altillimity/SatDump/actions/runs/3283808545

cosmopotamian, 2022-10-19:
> Can you try using this build? The latest beta is a bit outdated by this point, quite some things have been addressed in the meantime. https://github.com/altillimity/SatDump/actions/runs/3283808545

Wait! This build already fixed the decoding issue that was opened separately! Thank you!

I will checkout the gain setting when I do the test tomorrow! I will need to setup my antennas and everything again and will let you know. Looking forward!

Aang23, 2022-10-19:
OH! Well that's nice. I will wait to see if you've got issues tomorrow!

cosmopotamian, 2022-10-20:
> OH! Well that's nice. I will wait to see if you've got issues tomorrow!

Ok, so false alarm! This version still does not resolve the crash issue that happens after capturing or even offline decoding captured files on the system I have been using. Does not work on its original Windows setup nor on the virtualized windows (on the same hardware). But surprisingly, this version and even the beta worked well on another windows system I have (which also uses Windows 11 - like all others I have on me). So, it seems it is possibly, just possibly, hardware related, since the only difference between these two systems is the hardware, even though the virtualized one has virtualized hardware that is supposed to work differently and uses different Windows drivers. So, could it be related to CPU/Instruction sets/Extensions/etc? Pretty strange to me honestly.

On another note, regarding the gain slider, it does work well this time right off the start. So, if I slide it all the way up and I click start, the response is strong enough, reflective of the higher gain value. HOWEVER, I noticed that if I tick the AGC checkbox and untick it again, while the gain slider is already all the way up, it won't pick up on the gain value again. So, the issue resurfaces after ticking and unticking that AGC box.

Hope that helps!

Love your work, btw! Keep it up!

Aang23, 2022-10-20:
Interesting, so that's the same librtlsdr bug... Pretty easy fix, means I *absolutely* need to link AGC and gain settings together.

Basically, there's a bug where setting the AGC will reset gain.

Aang23, 2022-10-20:
Should be addressed now.

closed 2022-10-25 by Aang23

cosmopotamian, 2022-10-26:
> Should be addressed now.

Thank you! I will test it out and let you know if anything goes wrong.

cosmopotamian, 2022-10-27:
So I tried this, and sorry for reporting after you already closed the case.

I noticed that when the ADC is checked, sliding the gain left and right does affect the overall result, but then AGC effect kicks in again when you stop sliding, and the gain gets back to the highest (as set by the AGC option). Meaning that AGC checkbox is doing its job, but still being affected by the slider. Ideally, slider should have no effect on the gain, not even momentary.

No biggie, but just thought of throwing in my two cents on this as this is my impression from using the few other SDR applications.

commit 63269c26 (2022-10-20): RTL-SDR issues... Fixes #156