SatDump/SatDump #157: Saving last session settings when starting Satdump again.

cosmopotamian, 2022-10-19:
Nothing major!

I am doing extensive testing on GOES-R HRIT and realized that it would be amazing to save the last session's settings when starting satdump-ui again. Things like the source or choice and all relevant settings such as frequency, gain, checkboxes, and so on. I am especially interested in the Recording tab, but that obviously applies to all other tabs.

Aang23, 2022-10-19:
Actually, this is already partially being done now. Source settings (gain, samplerate, etc) are being saved now, as well as FFT settings.
Though, I guess frequency could be saved as well.

The actual source selection could also be done, but I may have to store the full ID for that, as there is no real guarantee about the "order" of sources in the list.

ZbychuButItWasTaken, 2022-10-20:
Keep in mind that while we are working on that it will not be implemented for everything. I'd keep it to only SDR settings and frequency, as everything else changes with pretty much every image and every pass.

cosmopotamian, 2022-10-20:
> Keep in mind that while we are working on that it will not be implemented for everything. I'd keep it to only SDR settings and frequency, as everything else changes with pretty much every image and every pass.

Would be amazing if you could keep the pipeline source as well, even if it takes saving the full ID as Aang23 mentioned. Also, hopefully the fft max and min levels. So upon the next start, like SDR++, you have the frequency, radio of choice and all other settings up there, and all you do is just click play and you see the spectrum centered around the same area with similar max and min levels.

Aang23, 2022-10-20:
That's definitely possible! Pre-Selecting what was selected on exit. I do see the incentive especially for xRIT / GEO setups.

For LEO users, a bit less useful but it could surely be appreciated as well, and shouldn't bother anyone anyway.

Thoughts @ZbychuButItWasTaken? If that sounds OK to everyone, could implement it rather easily. I'd go with just a quick search when opening, so in case the pipeline doesn't exist etc would just default back to normal.

Aang23, 2022-10-20:
Perhaps also save last pipeline parameters, such as dc block.

ZbychuButItWasTaken, 2022-10-21:
Yeah that would be ok. However I am quite against saving the last used pipeline. A favourite system I've been working on would solve it too pretty much (requiring just 1 more click to launch processing), while being much more LEO user friendly (which I'd consider most users to be).

Making the last pipeline save also would make the favourite system completly useless as when selecting last used the searchbox would scroll down to it, making the favourite invisible on launch... Which defeats the purpose.

ZbychuButItWasTaken, 2022-10-24:
Well I suppose that's done now, @Aang23 ?