SatDump/SatDump #162: Issues installing following SatDump READMe.md

Kyynetic, 2022-10-26:
Here is the error produced when cmake is ran...
[image]

Here is the src-core/libs folder with jpeg present...
[image]

ZbychuButItWasTaken, 2022-10-26:
Looks like you haven't followed the readme correctly.
`sudo apt install git build-essential cmake g++ pkgconf libfftw3-dev libvolk2-dev libjpeg-dev libpng-dev # Core dependencies. If libvolk2-dev is not available, use libvolk1-dev`

Aang23, 2022-10-26:
Yep... SatDump bundles its own (modified) libjpeg to decompress some J2K variants. Actually I should probably drop the dep on libjpeg.

Aang23, 2022-10-26:
Just removed the dependency, instead using the bundled one exclusively... I *have* to bundle it anyway as libjpeg-turbo on most systems does not have the required features (reading from a buffer, J2K, etc).

closed 2022-10-26 by Aang23

Kyynetic, 2022-10-26:
I removed and recloned the satdump repository after your change and the cmake command works now much thanks!

Aang23, 2022-10-26:
You're welcome!