SatDump/SatDump #163: GOES-R GRB crashes when used with live processing

Lmello1, 2022-10-26:
GOES-R GRB crashes seconds after starting live decoding. Multiple sources produce the same result.

ZbychuButItWasTaken, 2022-10-26:
That's very weird, could you provide some more information? Mostly about your system (os, hardware and SDRs used), but also logs will be very useful!

Aang23, 2022-10-26:
Hmmmm. Interesting about GRB. Does it work from a baseband though?

Aang23, 2022-10-26:
Just did a quick check. Runs here. Maybe yet another Windows issue?
[image]

Lmello1, 2022-10-27:
OS: Windows 10
crashes on laptop and desktop with hackrf, RTL-SDR, and even trying to decode from a blank file.
Logs do not even show that the decoding has started, however I can upload one if you would like

Aang23, 2022-10-27:
Fixed in https://github.com/altillimity/SatDump/commit/254b03b36971fc8efc66f773e91901ec9ca0ffec

closed 2022-10-27 by Aang23