SatDump/SatDump #166: Projection on METEOR doesn't work as expected

RAD750, 2022-10-28:
**Description of the issue**

The projection feature, when used on data received from METEOR satellites, has a weird behavior. Sometimes, it doesn't let you project at all (checkbox cannot be checked), but in my latest capture that I got ~10 minutes ago from METEOR-M N2-2, it would appear that instead, it just projects to a completely wrong location under South Africa, despite the pass clearly depicting Italy (attaching a screenshot). I'm not sure if this is a fault with SatDump or rather the RTC on Meteor-M acting up (again) but thought I'd report it.

This also happens on LRPT, and Meteor-M N2, although as of now I don't have a good .s file to test on (the pass of this morning was horrible) 

**Hardware (SDR/PC/OS)**
Thinkpad X260 (Intel i5-6200u, Intel HD Graphics 520); PlutoSDR

**Version (Eg, 1.0.0, CI Build #171)**
1.0.0 built the 24th of October 2022 (should I rebuild?)

**Screenshots**

Wrong projection for today:
[image]

No projection at all for yesterday's pass:
[image]

**CADUs and other data useful for debugging**

[Output from SatDump for today's pass](https://regulus.a-centauri.com/filedump/HRPT/METEORN2-2/202210281500_METEORN22.zip)

[Output from SatDump for yesterday's pass](https://regulus.a-centauri.com/filedump/HRPT/METEORN2-2/202210271600_METEORN22.zip)

[satdump.log](https://github.com/altillimity/SatDump/files/9888850/satdump.log)

Thank you :smile: :heart:

Aang23, 2022-10-28:
Oh and about M2... LRPT crashed again, but otherwise I believe things work as expected now.

Aang23, 2022-10-28:
PS : Hopefully you don't mind that I copied your "Files useful for debugging (CADU, etc)" part in the just-update bug report template :-)

closed 2022-10-28 by Aang23

RAD750, 2022-12-02:
Reopening this, because I couldn't get this evening's METEOR-M N2 pass to project, or to have the country outline (which is really useful on IR). Checkbox is disabled, it doesn't generate the low resolution projection.

Didn't try LRPT as I had just arrived home, so just had the time to grab the necessary for HRPT :joy: 

[baseband here](https://proxima.a-centauri.com/stuff/baseband_1700000000Hz_18-17-19_02-12-2022.wav)