SatDump/SatDump #167: FengYun-4 LRIT/HRIT Support

Aang23, 2022-10-28:
Support the FengYun-4x line of satellite downlinks :
- [x] LRIT
- [ ] HRIT-I
- [ ] HRIT-II (same freq as HRIT-III)
- [x] HRIT-III

Decoding process : 
- [x] DVB-S2 Demodulation
- [x] DVB-MPE UDP CADU Demux
- [ ] LRIT Files processing (+ DES decryption? Only if users have a valid key...)

LRIT is supposed to be fully encrypted. HRIT apparently contains some unencrypted data.

@sam210723 @Foxiks

Aang23, 2022-10-28:
@Foxiks actually, on the same frequency as HRIT-III you obtained, there is HRIT-II. I'm not sure of the specifications, but maybe it has different modulation parameters? There should also be HRIT-I on another frequency.

Foxiks, 2022-10-28:
unfortunately this satellite is at 3 degrees of elevation and gives a weak SNR for HRIT (excellent for LRIT). Also, this signal is not transmitted constantly and often turns off. I have been watching this. I did see a transmission on that frequency, but it was several times faster than the HRIT I gave you. But that fast transmission had an SNR of less than 1 dB.

Foxiks, 2022-10-28:
I think @sam210723 could get HRIT with a good SNR :)

sam210723, 2022-10-29:
I can give it a go, I've received LRIT before on a 90cm dish at about 27° elevation. Do we know what frequency HRIT-I is on?

[image]

Aang23, 2022-10-29:
Unfortunately, not yet. But I'd assume scanning around as we know LRIT/HRIT-II&III's freqs should do the trick if it isn't extremely weak.

Foxiks, 2022-10-30:
@Aang23 I have a recording of an HRIT signal that is a 1-2 dB stronger than the one I sent you. Try with this: https://disk.yandex.ru/d/z5OozNZlHy_oxA

Aang23, 2022-10-30:
[image]

A bit better, but still noisy...

Foxiks, 2022-10-30:
hm... Yes(

Aang23, 2022-11-14:
Update : It turns out sometimes I'm a bit dumb :-)

So, LRIT is non-compliant, but a valid JPEG2000 bitstream is present, unencrypted in all the files I checked as of now... Just very offset compared to anything we should expect in xRIT.
[image]
[image]

This was decoded from a short sample @DJRaov recorded back 1-2 years ago. I have pushed everything, so I would highly recommend attempting to received and harvesting CADU to figure out the rest. While it should just be J2K, I find the images a bit too dark.

sam210723, 2022-11-14:
Are they using a weird bit-depth for the images? I'll see what I can setup here for long CADU captures.

Aang23, 2022-11-14:
In a way they are. They downlink it in 16-bits, but only the lower 8-bits contain any sort of useful imagery!

I think a small dish should be fine enough on this downlink.

Foxiks, 2022-11-15:
[image]
This does not save data on Win and Kali

Aang23, 2022-11-15:
Did the PID ever go green?

Foxiks, 2022-11-15:
hm. no. Between 0 and ~8000

Aang23, 2022-11-15:
How long did you wait for? CADUs are not always transmitted, waiting a while can be necessary until you see some.

Foxiks, 2022-11-15:
~2 min 10 sec

Aang23, 2022-11-15:
Yes, so wait at least 15 mins or such.

Foxiks, 2022-11-15:
Okay!

Aang23, 2022-11-15:
@Foxiks any luck?

Foxiks, 2022-11-15:
I haven't tried it yet, but I should try later.

eswnl, 2022-11-18:
Hi, just downloaded the latest 1.0.1 release.
When trying to decode FY-4A, it shows as GK-2A decoder at the top. Is this an error?
[image]

Aang23, 2022-11-18:
Yes, that's a mistake on my end... Forgot to rename it. I also see it crashed on Windows, so another thing to look at.

Aang23, 2022-11-18:
@eswnl just fixed this.

eswnl, 2022-11-18:
I can now process the ts streams. Thanks Aang23.
[image]

Might be worth putting ts = true for HRIT as well.

Aang23, 2022-11-18:
Indeed, let me do it. Though for HRIT we have not yet seen data in the clear.

Aang23, 2022-11-18:
@eswnl just did it.
Is all your LRIT data looking so dark? (And did you manage to get a full disk yet perhaps?)

eswnl, 2022-11-19:
Yes they are dark. Now testing LRIT, will this output images automatically when it detects the end? Also what is the freq option under "TS input"?
 
[image]

eswnl, 2022-11-19:
Right, just processed the CADU file and got the image. I made the image brighter.
[image]
[image]

eswnl, 2022-11-19:
Oh, just another thing, the program crashes when I click stop processing. Am using Airspy R2. But the CADU file is output correctly.

eswnl, 2022-11-19:
HRIT demodulation was a bit unstable. Could not get any CADU files.
[image]
Will continue to use the TBS5927 for HRIT.

sam210723, 2022-11-19:
Nice work John, what is your RF setup for this?

eswnl, 2022-11-19:
1.2m dish, Goes sawbird, patch feed, Airspy R2.
For HRIT, I have to use the TBS5927.

eswnl, 2022-11-19:
Is it worth adding a decimation function? Airspy 10MSps seems a bit excessive for LRIT/HRIT, is it possible to gain some extra SNR?

eswnl, 2022-11-20:
hmm, there is a ch1 visible on the LRIT at 10992x11200. But still using the low bit depth.
[image]

sam210723, 2022-11-20:
How often are they sent on LRIT?

eswnl, 2022-11-20:
Seems that ch1, ch9, ch12 are sent every 15min on the hour. Only ch1 is at 10992x11200.

eswnl, 2022-11-20:
Seems HRIT signals are not constant, they come at xx:00 to xx:17 and xx:35 to xx:42.

eswnl, 2022-11-20:
Ch12 and ch2 came in the clear on HRIT. But ch2 image was blank.
[image]
Looks like higher bit depth to me.

Aang23, 2022-11-20:
@eswnl There are 2 different HRIT downlinks. The xx:00 and xx:35 ones are not the same, so that may be why SatDump's demodulator did not lock.

Indeed, that HRIT image is definitely higher bit depth. Do you have the data on hand? (TS/CADU), looks like I'll need to fixup some  stuff to output it correctly... Good to know HRIT *does* have some stuff in the clear!

eswnl, 2022-11-20:
Ok, I'll send the TS streams. The TBS5927 did all the demodulation, not SatDump.
Another error, FY-4A HRIT decoder shows up as LRIT in the UI.

Aang23, 2022-11-20:
[image]

Indeed, HRIT contains full 10-bits imagery. LRIT appears to only be 4-bits!? I'm thinking this may be somewhat due to JPEG-2000, maybe I need to increase bit depth *before* the inverse wavelet transform...

In HRIT, I have also found several "fast scan" streams. I am probably not decoding them properly yet, but hard to reverse-engineer without... More data. I'm now dumping them to imagery though. They appear to be called "REGC / REGX / RUEC".

[image]

> Ok, I'll send the TS streams. The TBS5927 did all the demodulation, not SatDump. I only fed SatDump the ts. Another error, FY-4A HRIT decoder shows up as LRIT in the UI.
Actually this it not a mistake. LRIT/HRIT are basically identical, apart from the contents. Hence, the same decoder is used and works exactly the same way (apart from different bit depths, etc).

As a note, I have just pushed the code that processes HRIT and the other imagery correctly.

eswnl, 2022-11-21:
Thanks @Aang23 

REGC = China Regional
REGX = Regional
RUEC = ??

I'll see if I can get more HRIT data and any on 1681 MHz, although I've never seen a signal there before.

Aang23, 2022-11-24:
I've got no clue about RUEC honestly. There's pretty much no documentation on FY-4 xRIT on this side of the firewall... I've only worked off the (very) scarce docs from @sam210723 and reverse-engineering.

There are 3 HRIT "types", HRIT-II and III are the ones you have decoded as of now... So we still lack HRIT-I, which per some old documentation is said to be on 1681. But of course it's hard to be sure.

eswnl, 2022-11-25:
I checked to see any signal on 1681 MHz. I don't think it exists.
I also can't find any frequency listings for FY-4B, my feeling is that it doesn't use L-band.

Aang23, 2022-11-26:
Interesting. Then I wonder where HRIT-I has gone!
As far as I could find 4B should also have xRIT on L-Band, but who knows... There's so little documentation.

Did you try pointing at it?

eswnl, 2022-11-27:
Actually, I concluded that since HRIT-1 would overlap with HRIT 1679 MHz, I figured that it might be on a different polarisation. So I swapped the polarisation and found it at 1680 MHz. Wider signal and too weak to do anything with (for me).
[image]
I think it uses much higher data rate.

eswnl, 2022-11-27:
FY-4B is near GK-2A so somehow their frequencies would have to coexist as well.

Aang23, 2022-11-27:
Ok, that's definitely it. HRIT-I likely contains all instruments / channels in full resolution I'd assume. I've got no idea if it contains anything in the clear but that's always a possibility...

Perhaps L-Band is currently off on 4B? I've looked at some pictures of the S/C and it definitely appears to carry the same L-Band antennas as 4A.

Aang23, 2022-11-27:
Anyway, to do a little summary : 
- LRIT is fully supported and contains 3 channels every 15 mins. Always in the clear. Bit depth appears to be around 4-bits
- HRIT-II and HRIT-III contain *some* unencrypted channels at 10-bits (apparently, including one other VIS compared to LRIT, perhaps could allow composites?). Full disks are fully supported, partial scans still need to be figured out properly.
- HRIT-I does exist, but un-decoded at the moment.

Aang23, 2022-11-27:
I've now made them able to live-process to imagery.
[image]

eswnl, 2022-11-28:
Tried pointing at FY-4B, all I found was this:
[image]

Aang23, 2022-11-28:
Interesting. Though that's more likely FY-2x telemetry or such.

sergsero, 2023-05-06:
Hello,

I am trying to make a long session of receiving HRIT signals FY4A at the frequency of 1679 MHz. Unfortunately, after switching the transmission from a broadband signal (~6 MHz) to a narrowband signal (~1.5 MHz), there is no repeated automatic recognition of the narrowband signal by the demodulator.

[image]

[image]

[image]

You have to manually perform the stop-start procedure with the demodulator button.

[image]

Is there any way to solve this problem?

Interestingly, sometimes the transmission of the image is interrupted in the current narrowband session by switching to broadband. And the missing part of the image is transmitted already in the next time narrowband mode.
Now, due to the forced manual break (stop/start demodulator) of the session, it is not possible to get a whole image.

[image]

[image]

pluggerlockett, 2023-06-23:
Morning all. @eswnl @sergsero can you please provide me some information regarding transmission times for FY-4A HRIT? I'd like to take another swing at it. Historically I've been able to get sync with my 1.2m offset but I never received any data. I'm also happy to point at FY-4B for testing too, both are easily visible from mine.

Aang23, 2023-08-02:
@pluggerlockett @eswnl @sam210723 did any of you ever collect more data from FY-4A to finish things up? And FY-4B to add pipelines? I wouldn't mind closing this if it can be considered done.

pluggerlockett, 2023-08-07:
I looked at both FY-4A and FY-4B on Saturday evening with my 1.2m offset to attempt to add more input @Aang23. FY-4A seems to be operating nominally. LRIT is still on the same schedule, and HRIT was observed transmitting in the following timeslots

- 00:00 to 00:16

- 00:32 to 00:40

Sadly I didn't receive any unencrypted data despite having what I would expect to be a good enough SNR.

[image]

I then attempted to look at FY-4B with less than stellar results. The LRIT signal has changed significantly, last time I looked they had pulled back the power output to not interfere with GK-2A HRIT. But on Saturday it was basically a stonking huge carrier and not much else.

[image]

Honestly, I'm not sure what to make of that. I then attempted to look at FY-4B HRIT, in nearly 30 minutes I didn't see anything transmit, and I was looking from ~00:20 to 00:48.

I hope that helps?

nil-coder-sudo, 2026-10-26:
I try to crack with Hachcat,but I failed, Anyone Have key?

nil-coder-sudo, 2026-10-26:
[image]
But I think it will take a long time

lzh173, 2026-10-26:
> 30\. 我试着用Hachcat破解，但失败了，有人有钥匙吗？

you can see this https://github.com/SatDump/SatDump/issues/1157

nil-coder-sudo, 2026-11-02:
But it can't change the crack time, it need very very long too.....