SatDump/SatDump #170: Output channels to separate subdirectories

JVital2013, 2022-10-30:
**What would you want to be added to satdump?**
When decoding imagery from satellites with multiple channels, save images from each channel to its own subdirectory. This can be a global option, or an option configured within the pipeline JSON files.

**Describe the feature you'd like to have**
For example, when decoding GOES HRIT images, the current directory structure for ABI images looks like this:
```
Output
└───IMAGES
    ├───GOES-16
    │   ├───Full Disk
    │   │       G16_13_20221030T130020Z.png
    │   │       G16_13_20221030T133020Z.png
    │   │       G16_14_20221030T123020Z.png
    │   │       G16_14_20221030T130020Z.png
    │   │       G16_2_20221030T130020Z.png
    │   │       G16_2_20221030T133020Z.png
    │   │       G16_7_20221030T130020Z.png
    │   │       G16_7_20221030T133020Z.png
    │   │       G16_8_20221030T130020Z.png
    │   │       G16_8_20221030T133020Z.png
    │   │       G16_9_20221030T130020Z.png
    │   │       G16_9_20221030T133020Z.png
    │   │       G16_FC_20221030T130020Z.png
    │   │       G16_FC_20221030T133020Z.png
    │   │       
    │   ├───Mesoscale 1
    │   │       G16_13_20221030T125224Z.png
    │   │       G16_13_20221030T130724Z.png
    │   │       G16_2_20221030T125224Z.png
    │   │       G16_2_20221030T130724Z.png
    │   │       G16_7_20221030T125224Z.png
    │   │       G16_7_20221030T130724Z.png
    │   │       G16_FC_20221030T135224Z.png
    │   │       G16_FC_20221030T140724Z.png
    │   │       
    │   └───Mesoscale 2
    │           G16_13_20221030T125254Z.png
    │           G16_13_20221030T130754Z.png
    │           G16_2_20221030T125254Z.png
    │           G16_2_20221030T130754Z.png
    │           G16_7_20221030T125254Z.png
    │           G16_7_20221030T130754Z.png
    │           G16_FC_20221030T135254Z.png
    │           G16_FC_20221030T140754Z.png
    │           
    └───GOES-18
        └───Full Disk
                G18_13_20221030T125021Z.png
                G18_13_20221030T135021Z.png
```

In my ideal world, they will look like this:

```
Output
└───IMAGES
    ├───GOES-16
    │   ├───Full Disk
    │   │   ├───ch02
    │   │   │       G16_2_20221030T130020Z.png
    │   │   │       G16_2_20221030T133020Z.png
    │   │   │
    │   │   ├───ch07
    │   │   │       G16_7_20221030T130020Z.png
    │   │   │       G16_7_20221030T133020Z.png
    │   │   │
    │   │   ├───ch08
    │   │   │       G16_8_20221030T130020Z.png
    │   │   │       G16_8_20221030T133020Z.png
    │   │   │
    │   │   ├───ch09
    │   │   │       G16_9_20221030T130020Z.png
    │   │   │       G16_9_20221030T133020Z.png
    │   │   │
    │   │   ├───ch13
    │   │   │       G16_13_20221030T130020Z.png
    │   │   │       G16_13_20221030T133020Z.png
    │   │   │
    │   │   ├───ch14
    │   │   │       G16_14_20221030T123020Z.png
    │   │   │       G16_14_20221030T130020Z.png
    │   │   │
    │   │   ├───ch15
    │   │   │       G16_15_20221030T123020Z.png
    │   │   │       G16_15_20221030T130020Z.png
    │   │   │
    │   │   └───fc
    │   │           G16_FC_20221030T130020Z.png
    │   │           G16_FC_20221030T133020Z.png
    │   │
    │   ├───Mesoscale 1
    │   │   ├───ch02
    │   │   │       G16_2_20221030T125224Z.png
    │   │   │       G16_2_20221030T130724Z.png
    │   │   │
    │   │   ├───ch07
    │   │   │       G16_7_20221030T125224Z.png
    │   │   │       G16_7_20221030T130724Z.png
    │   │   │
    │   │   ├───ch13
    │   │   │       G16_13_20221030T125224Z.png
    │   │   │       G16_13_20221030T130724Z.png
    │   │   │
    │   │   └───fc
    │   │           G16_FC_20221030T135224Z.png
    │   │           G16_FC_20221030T140724Z.png
    │   │
    │   └───Mesoscale 2
    │       ├───ch02
    │       │       G16_2_20221030T125254Z.png
    │       │       G16_2_20221030T130754Z.png
    │       │
    │       ├───ch07
    │       │       G16_7_20221030T125254Z.png
    │       │       G16_7_20221030T130754Z.png
    │       │
    │       ├───ch13
    │       │       G16_13_20221030T125254Z.png
    │       │       G16_13_20221030T130754Z.png
    │       │
    │       └───fc
    │               G16_FC_20221030T135254Z.png
    │               G16_FC_20221030T140754Z.png
    │
    └───GOES-18
        └───Full Disk
            └───ch13
                    G18_13_20221030T125021Z.png
                    G18_13_20221030T135021Z.png
```

**Describe alternatives you've considered**
A script could be written to run after-the-fact to reorganize the files, but "tack-ons" like this can be less reliable. It would be nice to make this configurable to meet the end-user's needs.

Another alternative is to make the directory tree completely customizable by the end user by specifying the directory structure in the pipeline json file.

**Additional context**
When viewing the images that are output, it would be nicer if all the images in a given directory are homogenous. It can make them easier to write scripts to parse them as well.

ZbychuButItWasTaken, 2022-10-30:
Hmmm I can see the use for that, but implementing that would mean rewriting the saving part of every decoder out there.. In my opinion filenames are descriptive enough and are easy enough go be separated by whatever the next processing step is.

Aang23, 2022-10-30:
@ZbychuButItWasTaken I agree, definitely NOT to implement on all satellites. All LEOs definitely not. Though on GEOs, sub-dividing more can sure be handy with how much data this actually is over even a few days.

@JVital2013 what would you think of a slightly different way? Classify not per channel, but by scan timestamp?
So :
ABI_TIMESTAMP : 
- Ch1
- Ch2
- Ch3
This would match the scheme utilized on GRB and GVAR, and make a lot of sense to drop in a product.cbor in each directory.

JVital2013, 2022-10-31:
I could see the utility in classifying it by timestamp instead of channel to some degree, and it probably makes sense depending on how you use SatDump.

At least on HRIT, some products are offset from each other in time. For example, all the non-CMIP imagery comes down at odd times that does not line up with the other full disk scans. With your suggestion, would they have their own folder with one product in them?

For my uses, I see that breakdown as causing more problems than it solves. I'm more interested in how a single product changes over time, not what all products look like at a given time. This might not be representative of all SatDump users, though.

Thinking more about this: no matter how it's changed, making changes like this would probably upset somebody's workflow/use case. If SatDump went to a model like goestools's goesproc, where the output folder structure is completely customizable, I could see it potentially working. Make the default pipeline json have a config that matches the current directory structure, but then users can change the directory structure if they're particular pretentious about it (like me, apparently).

Ehhh, sounds complicated. I won't take it personally if this idea needs to be scrapped

Aang23, 2022-11-01:
In this instance, already-processed products would be in its directory, separate, as it's not raw imagery SatDump could then process further (which is what we care about - As SatDump is then supposed to load all available channels and produces several products, potentially similar to what GOES also downlinks if the user wishes).

I do agree with your conclusion though... It's probably a simpler and better idea to make it configurable. It's about as much work for me as changing the current structure so I don't really care :-)

I will likely ship what I explained earlier as a default, but this way you will be able to change it to your liking, and I assume to work with Vitality GOES (if you need anything else in that regard for compatibility / etc, feel free to ask).

Chances are I will go with something such as 
`{channel_name/product_name}/{utc_timestamp}/xxxx.png`
Where each placeholder would then be replaced.

JVital2013, 2022-11-01:
Sounds great, thanks! Your example configuration is exactly what I was thinking.

I appreciate you considering these suggestions. Implementing this request would result in 100% Vitality GOES compatibility; the request about selectable file formats is not needed, just a nicety. I have some local code I was testing that would make Vitality GOES compatible with SatDump as-is, but I might hold on that until I see what happens here.

Aang23, 2022-11-01:
I was planning on rewriting a large part of the xRIT satellites (including GOES) to be more generic soon.
As SatDump can process data for several downlinks using the format : 
- GOES-N LRIT
- GOES-R HRIT
- GK-2A LRIT/HRIT
- ELEKTRO-L LRIT/HRIT
- HimawariCast (Also uses HRIT files)
- Soon enough FengYun-4x LRIT/HRIT
Nearly all the code before getting down to xRIT for all those is identical, and quite a large part of the code to recompose full-disk imagery and so on is also extremely similar. I will be combining most of the codebase soon, which will involve a decent rewrite. 
Hence, implementing this suggestion at the same time really won't be a problem, and chances are I will be keeping the same system for all GEOs. Perhaps that could allow you to support non-GOES satellites as well on your end if you wish?

JVital2013, 2022-11-02:
That's the goal - it's the biggest reason why I want to make sure Vitality GOES is compatible with SatDump. If SatDump is configurable and consistent, my software can be made to support it and all the satellites under the same umbrella.

creinemann, 2022-11-03:
I'll just chime in:  the output file structure by channel rather than by Full Disk, Meso 1 Meso 2 like this sure would make my Sanchez scripts easier to run with satdump output....
Output
└───IMAGES
    ├───GOES-16
    │   ├───Full Disk
    │   │   ├───ch02
    │   │   │       G16_2_20221030T130020Z.png
    │   │   │       G16_2_20221030T133020Z.png
    │   │   │
    │   │   ├───ch07
    │   │   │       G16_7_20221030T130020Z.png
    │   │   │       G16_7_20221030T133020Z.png
    │   │   │
    │   │   ├───ch08
    │   │   │       G16_8_20221030T130020Z.png
    │   │   │       G16_8_20221030T133020Z.png
    │   │   │
    │   │   ├───ch09
    │   │   │       G16_9_20221030T130020Z.png
    │   │   │       G16_9_20221030T133020Z.png
    │   │   │
    │   │   ├───ch13
    │   │   │       G16_13_20221030T130020Z.png
    │   │   │       G16_13_20221030T133020Z.png
    │   │   │
    │   │   ├───ch14
    │   │   │       G16_14_20221030T123020Z.png
    │   │   │       G16_14_20221030T130020Z.png
    │   │   │
    │   │   ├───ch15
    │   │   │       G16_15_20221030T123020Z.png
    │   │   │       G16_15_20221030T130020Z.png
    │   │   │
    │   │   └───fc
    │   │           G16_FC_20221030T130020Z.png
    │   │           G16_FC_20221030T133020Z.png
    │   │
    │   ├───Mesoscale 1
    │   │   ├───ch02
    │   │   │       G16_2_20221030T125224Z.png
    │   │   │       G16_2_20221030T130724Z.png
    │   │   │
    │   │   ├───ch07
    │   │   │       G16_7_20221030T125224Z.png
    │   │   │       G16_7_20221030T130724Z.png
    │   │   │
    │   │   ├───ch13
    │   │   │       G16_13_20221030T125224Z.png
    │   │   │       G16_13_20221030T130724Z.png
    │   │   │
    │   │   └───fc
    │   │           G16_FC_20221030T135224Z.png
    │   │           G16_FC_20221030T140724Z.png
    │   │
    │   └───Mesoscale 2
    │       ├───ch02
    │       │       G16_2_20221030T125254Z.png
    │       │       G16_2_20221030T130754Z.png
    │       │
    │       ├───ch07
    │       │       G16_7_20221030T125254Z.png
    │       │       G16_7_20221030T130754Z.png
    │       │
    │       ├───ch13
    │       │       G16_13_20221030T125254Z.png
    │       │       G16_13_20221030T130754Z.png
    │       │
    │       └───fc
    │               G16_FC_20221030T135254Z.png
    │               G16_FC_20221030T140754Z.png
    │
    └───GOES-18
        └───Full Disk
            └───ch13
                    G18_13_20221030T125021Z.png
                    G18_13_20221030T135021Z.png

Aang23, 2022-11-17:
Noted @creinemann, thanks for your input!

Now that the first "part" of the rewrite of xRIT I had planned is done & released, I'm expecting to implement the 2nd part (including this) for the next release.

JVital2013, 2023-08-24:
After letting this sit for 9 months, I wanted to chime back in since my opinion has changed slightly. My change of heart has nothing to do with the fact that I made Vitality GOES dynamic enough to handle most SatDump output. No, that must be unrelated :smile:

I'm no longer that interested in this change personally. I know you want to push a new release soon, and I have no problem putting this off for a while after this release. That's scheduled for the same day Half-Life 3 comes out, right? :smile:

I want to keep the issue open since I think this is working on this at some point is still worthwhile. #284 is a similar thought, and I had a Goestools for Windows user ask me how to save EMWIN data to a specific folder for another piece of 3rd-party software to process, so I know there's still interest in this type of thing. I just don't think it's a priority.

Aang23, 2023-09-14:
My idea was (and still is) to take of this as a part of a wider "rework" of the GEOs. Right now, they're handled bypassing the products system entirely which is not desirable in the long run. 
The idea later is to have them handled (for raw image channels) just like every LEO instrument and have it usable directly in the viewer, satdump_cfg.json for composites/projections and so on.

Since the problems to be overcome for (nearly) all GEOs are very similar, I am planning on doing a common thing for "xRIT" links, etc. 

However... Quite a bit of work and there's been more important quite a lot. I'll probably re-consider after the next release or such, first the products aspect, and perhaps at the same time said custom paths, etc.

closed 2024-04-23 by Aang23