SatDump/SatDump #171: Make output image file format selectable

JVital2013, 2022-10-30:
**What would you want to be added to satdump?**
Allow the user to customize the default output image format for images rendered by SatDump. Can be a global setting.

**Describe the feature you'd like to have**
When SatDump renders an image, it prefers to save it as a png. This should be a user setting to allow the user to select jpg, png, bmp, and others as you see fit.

PNG files are the obvious best choice to preserve quality and transparency, but other file formats may be more desirable in some contexts.

**Describe alternatives you've considered**
File format can be changed by a secondary script that "picks up" decoded files and converts them to the desired format, but this seems like "extra work" when SatDump itself could handle it.

**Additional context**
If a satellite downlinks already-rendered files (ex. GOES EMWIN images), this setting should be ignored and the original file should be saved. I see this setting as only applying to images that need to be rendered by SatDump.

Aang23, 2022-10-30:
I'd assume you would want to use JPEG or something similar for GOES imagery to save on space? Though, I'd guess it would indeed be a good idea to make it an option overall, or perhaps override-able with a pipeline parameters.

Thoughts @ZbychuButItWasTaken?

ZbychuButItWasTaken, 2022-10-30:
Agreed, but def not by default. More options is always better!

JVital2013, 2022-10-31:
Correct, mostly for disk space savings. Obviously a low-priority suggestion!

JVital2013, 2023-08-03:
@Aang23 @ZbychuButItWasTaken I'm comfortable enough with the code base at this point that I'm working on this myself. I'll submit a PR in a day or two, but I have a working copy over at https://github.com/JVital2013/SatDump/tree/FixNewNameSave.

Overview of changes:
 - Allows the user to interactively choose between jpg and png when saving imagery in the viewer/projector
 - Adds a setting that allows users to choose between png and jpg for all automated processed image saves (defaults to png)
 - Extends `save_img` to check if an extension was passed in the file name. If not, it will use the extension set in settings. Also added the ability to pass along the `fast` parameter to `save_png` if necessary
 - Modifies `ImageProducts` to optionally accept images without a file extension, and use the extension configured in settings if unspecified.
 - Move `logger->info()` for saving an image into `save_img` to greatly reduce repeated logging code, and standardize the output across plugins
 - Updates all the plugins to use `save_img` instead of `save_png`, and pass the file without an extension to allow the system setting to take effect. _This is still a WIP; y'all have written a lot of plugins..._

This results in a fairly flexible implementation, that allows overriding at the plugin/pipeline level if necessary. Just pass a .png file extension, and `save_img` will force save it as a png.

Other than finishing the plugins, I'm also going to look at changing the JPEG quality. Right now it saves them at 75, which might be a bit compressed. Upping it to 80-90 is probably a good idea. Although - a full-res GOES GVAR image is only 40 MB right now!

**Setting:**
[image]

closed 2023-08-06 by Aang23