SatDump/SatDump #172: Automatic boundary coordinates detection for projections

RAD750, 2022-11-01:
**What would you want to be added to SatDump?**

I think it would be a good idea to have a button that could automatically project the image ONLY in the world area where there actually is satellite imagery available. 

For polar satellites direct broadcasting, it would not make sense to project the image on the whole world, as it wastes memory space. It makes sense for things like GAC or geostationary satellites, but for all other uses, not so much.

**Describe the feature you'd like to have**
Ideally, a button "Automatically detect coordinates" in the projection panel, would automatically detect where the vertices of the projected images fall, so they can be projected without either guessing the vertices by trial and error, or projecting the whole world and cutting using an image manipulation tool.

**Describe alternatives you've considered**
All alternative solutions are completely manual: either guess the coordinates to put in the equirectangular projection, or project the whole world to a very large size (10000x5000) and then cut using GIMP or similar

Thanks as always guys :heart:

ZbychuButItWasTaken, 2022-11-01:
That definitely could be added np. Meanwhile, I would recomend using stereographic projection centered at your location. That should work for all the LEO images and not take up much additional space.

RAD750, 2022-11-01:
That's what I do (I have to select a location 3-4 degrees of latitude lower than mine because I only get the south due to hills) for now :+1:

ZbychuButItWasTaken, 2022-11-01:
Hmm thinking on the technical side it is not that simple. How do you imagine satdump should calculate the size of the output image? With the current setup it would not be easy to do at all.

ZbychuButItWasTaken, 2022-11-09:
hmmmmmmmm came back to this thinking how to do it and I'm not sure what the best way would be.
Issue is, that the output image size is selected by the user, so that shouldn't really change on its own. Besides that, even if we were to edit it, how would it determine what size to use for the projection (size of the entire map)? Can't calculate that from nothing. 

@Aang23, do you have any ideas for this one?

Aang23, 2022-11-24:
Ah, missed this.
It's possible, but will still be a bit tricky : 
1) We pretty much need to project at least some of the imagery to know the boundaries. We don't know where things are before they're geo-located. So having such an option will take a bit of time.
2) Not all projections are ""square"". Hence, apart from equirectangular a somewhat complicated algorithm needs to determine the parameters necessary to cover everything. Not trivial, but possible
3) This would only work reliably with good data.

I have a feeling a "warp" option would be better suited than supporting all projection algos.

RAD750, 2022-11-24:
What I was thinking was something akin to what WXtoIMG does:
[image]

The easiest solution would be, I think, to produce such images for the automatically generated projections when the pipeline is done (no need to produce a huge world image unless dealing with GAC, for those), so the user can easily combine them using GIMP or another graphics tool, especially if the background is either solid black or transparent.

It is not required, for me at least, to produce automatically cropped images when dealing with the viewer, as the user could very well need to adjust other parameters and so it is not a big deal to also center the projection manually when you are done with the rest like map layers etc.

But for the automatically generated projections, it just seems to me like a waste of memory to produce them the way they are now, because even for GAC or geo sats, the resulting image will never cover the whole world. And usually, for polar sats, the resulting projection is of a so low resolution that it is almost useless as it is generated. :)

Aang23, 2022-11-24:
Oh, so this was purely about auto-generated images? Then it's a bit of different story. Those are purely meant to be "quick-look". Basically those you open up to see roughly what you got coverage-wise. They're just low-resolution by design. 

I could consider switching those quick-looks over to the raw "warped" image perhaps, but this is already something you can do in satdump_cfg.json. Projections that are generated are defined there so you could edit the area covered to only be your station, which would achieve what you want without the added complexity.

The initial idea was that you would load up the images in SatDump's viewer (products), and project / mix them using the projection tool to get full-resolution composites. If you intend to do that anyway in Gimp, perhaps the tools already in SatDump could cover your usecase? 
Personally, I often use my station's location in Stereo projection mode scale 2. In my own configuration I have that automatically-generated.
Here are a few samples (from older versions though, and imperfect data. Just what I had on hand). Note I had ASCAT overlaid on those :
[image]
[image]

Aang23, 2022-12-08:
Projections settings are now saved, which should cover this issue as well.

closed 2022-12-08 by Aang23