SatDump/SatDump #174: ld: library not found for -lomp

benb0jangles, 2022-11-04:
Any idea?

Compile error on MacOs Catalina 10.15.7

Latest satdump (as of November 4th 2022) version https://github.com/altillimity/SatDump

`after make -j4:`
error:
```
[ 51%] Building CXX object src-core/CMakeFiles/satdump_core.dir/spdlog/file_sinks.cpp.o
[ 51%] Building CXX object src-core/CMakeFiles/satdump_core.dir/spdlog/fmt.cpp.o
[ 52%] Building CXX object src-core/CMakeFiles/satdump_core.dir/spdlog/spdlog.cpp.o
[ 52%] Building CXX object src-core/CMakeFiles/satdump_core.dir/spdlog/stdout_sinks.cpp.o
[ 52%] Linking CXX shared library ../libsatdump_core.dylib
ld: library not found for -lomp
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libsatdump_core.dylib] Error 1
make[1]: *** [src-core/CMakeFiles/satdump_core.dir/all] Error 2
make: *** [all] Error 2

```

following build process:
```
# macOS: Install dependencies
brew install cmake volk libpng glew glfw nng pkg-config llvm libomp

# macOS ONLY: build and install libfftw3
# if you install fftw via brew, cmake won't be able to find it
wget http://www.fftw.org/fftw-3.3.9.tar.gz
tar xf fftw-3.3.9.tar.gz
rm fftw-3.3.9.tar.gz
cd fftw-3.3.9
mkdir build && cd build
cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_TESTS=false -DENABLE_FLOAT=true -DENABLE_THREADS=true -DENABLE_SSE=true -DENABLE_SSE2=true -DENABLE_AVX=true -DENABLE_AVX2=true ..
make
sudo make install
cd ../..
rm -rf fftw-3.3.9

# Finally, SatDump
git clone https://github.com/altillimity/satdump.git
cd satdump
mkdir build && cd build
cmake -DCMAKE_BUILD_TYPE=Release ..                             # MacOS
make -j4
```

Aang23, 2022-11-04:
I guess something changed in the homebrew package... Will have too boot my VM to check.

benb0jangles, 2022-11-05:
clang --version
Apple clang version 12.0.0 (clang-1200.0.32.29)
Target: x86_64-apple-darwin19.6.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin

benb0jangles, 2022-11-05:
brew install libomp:

libomp is keg-only, which means it was not symlinked into /usr/local,
because it can override GCC headers and result in broken builds.

For compilers to find libomp you may need to set:
  export LDFLAGS="-L/usr/local/opt/libomp/lib"
  export CPPFLAGS="-I/usr/local/opt/libomp/include"

benb0jangles, 2022-11-05:
after doing:
`brew install libomp`

```
export LDFLAGS="-L/usr/local/opt/libomp/lib"
export CPPFLAGS="-I/usr/local/opt/libomp/include"
```

fixed the problem

benb0jangles, 2022-11-05:
so after completing build with:
`sudo make install`
I run:
`./satdump-ui`

then I get:

[image]

benb0jangles, 2022-11-05:
Also I found that on MacOs Locally-installed software belongs in /usr/local, not /usr/share or other. Due to Mac security measures only /usr/local is allowed.

so storing pipelines in /usr/share/satdump/pipelines will not be allowed and compiling will fail. The Mac user must go through time consuming process to disable SIP security to fix, but it would be more easy if change:
/usr/share/satdump/pipelines 
to
/usr/local/satdump/pipelines

Aang23, 2022-11-07:
Interesting. Thanks for debugging! I will likely add those paths to CMake's and be done with it, should work OK.
Noted about the wrong install paths, I will change that. 
Thanks!

Aang23, 2022-11-07:
@benb0jangles the omp issue should be fixed.

Aang23, 2022-11-07:
I have also just changed the default install prefix to be /usr/local. Should be fine now!

closed 2022-11-07 by Aang23

benb0jangles, 2022-11-07:
Excellent! Thank you.