SatDump/SatDump #178: ELEKTRO L3 HRIT

sq7r0, 2022-11-06:
Hi

getting an error when trying to decode elektro l3 hrit using Satdump 1.0.0 beta on SDRPlay RSPdx, runs on Win11. Some parts are passing okay, even there is no enough signal, but at some point im getting this :

[11/06/22 - 22:12:58] (I) New LRIT file : H-000-GOMS3_-GOMS3_4_____-00_7_076E-000003___-202211062100-C_
[11/06/22 - 22:12:58] (D) This is image data. Size 2784x464
[11/06/22 - 22:12:58] (D) Wavelet Compression is used, decompressing...
[11/06/22 - 22:13:00] (I) Progress -nan(ind)%, SNR : 4.584462dB, Peak SNR: 5.987127dB
[11/06/22 - 22:13:00] (I) Progress -nan(ind)%, Viterbi : SYNCED BER : 0.120850, Deframer : SYNCED
[11/06/22 - 22:13:00] (I) Progress -nan(ind)%
[11/06/22 - 22:13:00] (I) Decompressing Wavelet...
Software Exception Trace (Source: D:\a\SatDump\SatDump\plugins\elektro_arktika_support\DecompWT\CWBlock.cpp, Line: 777)
Software Exception!  Error Code: 1, Error Message:
Software Exception Trace (Source: D:\a\SatDump\SatDump\plugins\elektro_arktika_support\DecompWT\CWBlock.cpp, Line: 790)
Software Exception Trace (Source: D:\a\SatDump\SatDump\plugins\elektro_arktika_support\DecompWT\CWTDecoder.cpp, Line: 59)
Software Exception Trace (Source: D:\a\SatDump\SatDump\plugins\elektro_arktika_support\DecompWT\CWTDecoder.cpp, Line: 154)
Software Exception Trace (Source: D:\a\SatDump\SatDump\plugins\elektro_arktika_support\DecompWT\CWTDecoder.cpp, Line: 305)
Software Exception Trace (Source: D:\a\SatDump\SatDump\plugins\elektro_arktika_support\DecompWT\CWTDecoder.cpp, Line: 487)
Software Exception Trace (Source: D:\a\SatDump\SatDump\plugins\elektro_arktika_support\DecompWT\CompressWT.cpp, Line: 45)

then the waterfall and constellation freezes and after a while whole satdump app too. No problem receiving and decoding L3 lrit

cheers
Krzysztof

Aang23, 2022-11-07:
Would you be able to provide a baseband on the next transmission? I will debug on windows. Looks like eumetsat's decompressor is the issue.

sq7r0, 2022-11-07:
@Aang23 sure, here you go : https://cloud.egmont.pl/index.php/s/nnsVFqFEOWVP4bJ
I know the signal is not perfect, but it stuck on 16% and freezes. Btw I have downloaded the newest windows build and its even worse. When i choose L3 HRIT and start to live decode it freezes on first bytes, like below.
[image]

Aang23, 2022-11-07:
@sq7r0 mind trying the latest commit? It should be fixed.

Aang23, 2022-11-07:
Full data : https://we.tl/t-dvdh1SaX7B

closed 2022-11-07 by Aang23