SatDump/SatDump #180: ELEKTRO-L TLM

sq7r0, 2022-11-07:
Hi,

another issue found on decoding the Elektro L3 telemetry. Whole app freezes just after start the process. No error seen in logs
[image]
Saving the baseband and pass offline processing creates cadu file only. Is that correct?

cheers
Krzysztof

Aang23, 2022-11-07:
Known issue. Do you have a baseband I could use to fix it up? Forgot where I put mines :-)

sq7r0, 2022-11-07:
@Aang23 you are fast :D havent finished to upload the file and got reply xd Here is the baseband : https://cloud.egmont.pl/index.php/s/xjptIsYUtg0YUZe

Aang23, 2022-11-07:
@sq7r0 check the latest commit once CI is done, should be fixed.

sq7r0, 2022-11-07:
@Aang23 seems its fine now, works stable :) Do you plan any interpreter of this data?
[image]

Aang23, 2022-11-07:
Yes, I do! I have mostly reverse-engineered it during a solar-flare last time to somewhat reliably identify things. Though, do keep in mind all you will get out of it is radiation monitoring. Still useful and interesting, but not imagery!

sq7r0, 2022-11-07:
@Aang23 yes thats perfect, it will be nice to combine with images taken with solar filter during flare eruptions, like this one i got some time ago ;)
[image]

Aang23, 2022-11-07:
Indeed! It's some pretty nice data still, especially during this sort of events...

Awesome image by the way!

closed 2022-11-07 by Aang23