SatDump/SatDump #188: demodulating verbose in satdump-ingestor?

bclswl0827, 2022-11-15:
**What would you want to be added to satdump?**
satdump-ingestor returns more data (decoder, deframer, progress, fec, lock, etc.) than just SNR
**Describe the feature you'd like to have**
**Describe alternatives you've considered**
**Additional context**
satdump-ingestor can be useful when used as a backend of unmanned ground station, it should return the same amount of data as satdump-ui

Aang23, 2022-11-15:
The ingestor is a not thing anymore. The feature is now in the main CLI app.
You may perhaps want to look into the stats server? That's a way for it to provide all status information over HTTP in a JSON format.

https://www.satdump.org/posts/basic-usage/#live-processing

bclswl0827, 2022-11-15:
Great! I'm still using the old version, thanks for the quick response!

closed 2022-11-15 by bclswl0827