SatDump/SatDump #190: Corrupt EMWIN images on Windows

JVital2013, 2022-11-21:
**Description of the issue**

When Receiving the HRIT signal from GOES-16 with SatDump on Windows, the EMWIN images are corrupt. All other data is fine, and there are no errors in the log (including no CRC errors). The issue is not present when running SatDump on Linux with the same hardware.

**Hardware (SDR/PC/OS)**

SDR: NooElec NESDR SmarTee XTR
OS: Windows 10 Build 21H2
CPU: Intel(R) Core(TM) i5-6600K CPU
Memory: 16GB RAM
GPU: AMD Radeon RX 580
Visual Studio 2015-2022 Redistributable version: 14.34.31931.0

**Version (Eg, 1.0.0, CI Build #171)**

Every build I tested, 1.0 through Windows CI Build #729

**Logs after the crash (satdump.logs)**

N/A

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

A bunch of examples of corrupt files: [EMWIN.zip](https://github.com/altillimity/SatDump/files/10052692/EMWIN.zip). I have these same files. 

Most of the images are completely corrupt, but some of them look ok for the first few blocks:
[image]

Aang23, 2022-11-21:
Hello! Well, this has actually already been fixed, after 1.0.0 and build #729.

Aang23, 2022-11-21:
Ah wait, I misread the build number.

Do you have a CADU file I could test with on hand? I know this is fixed on Linux, but perhaps I did miss something.

Aang23, 2022-11-21:
Well, found one. Will debug shortly.

JVital2013, 2022-11-21:
Sounds good - I can ship one over in a few hours too. I believe I still have the cadu file from the images in the attached zip

JVital2013, 2022-11-21:
Sorry, it looks like the SatDump live processing no longer outputs a cadu file automatically for me anymore. Can you point me in the right direction on how to get that?

Aang23, 2022-11-21:
You need to edit the GOES.json file in pipelines. In goes_hrit, change

```
 "default": [
                [
                    1,
                    0
                ],
                [
                    2,
                    0
                ],
                [
                    3,
                    0
                ]
            ],
```
to
```
 "default": [
                [
                    1,
                    0
                ],
                [
                    2,
                    0
                ]
            ],
```

Aang23, 2022-11-21:
As a note, I've just ran a CADU on my (Linux) machine. 
[image]
[image]

All images are fine. Time to test on Windows.

Aang23, 2022-11-21:
Fixed, this was an usual Windows issue (differentiating text & binary writing when it's the same on Linux)

closed 2022-11-21 by Aang23

JVital2013, 2022-11-21:
Thanks - unfortunately, SatDump just crashes now as soon as I start the GOES HRIT pipeline. Log attached
[satdump.log](https://github.com/altillimity/SatDump/files/10059866/satdump.log)

Aang23, 2022-11-21:
@JVital2013 fixed.

JVital2013, 2022-11-21:
@Aang23 indeed it is! I can confirm that EMWIN imagery now saves fine on Windows, and there are no more crashes with live processing.

Thanks for your hard work on this. SatDump is rapidly maturing into a highly-functional piece of code!

Aang23, 2022-11-21:
Awesome.

You're welcome! Thanks. Will be getting to your other issues soon by the way.