SatDump/SatDump #192: Himawaricast - stol argument out of range

eswnl, 2022-11-22:
**Description of the issue**
When trying to process himawaricast transport stream

**Hardware (SDR/PC/OS)**
TBS5927

**Version (Eg, 1.0.0, CI Build #171)**

**Logs after the cras (satdump.logs)**
[11/22/22 - 13:30:53] (I) Using input frames E:\DVB_stream\0.0E_4148.146_H_2586_(2021-05-03 02.58.45)_dump.ts
[11/22/22 - 13:30:53] (I) Decoding to C:\Users\john\Desktop\HimawariTest
[11/22/22 - 13:30:53] (I) New xRIT file : IMG_DK01B11_202105021750_004
[11/22/22 - 13:30:53] (D) This is image data. Size 2750x275
[11/22/22 - 13:30:53] (E) Fatal error running pipeline : stol argument out of range
[11/22/22 - 13:31:12] (I) Saving user config at settings.json
[11/22/22 - 13:31:12] (I) UI Exit
[11/22/22 - 13:31:12] (I) Exiting!

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
[image]

Aang23, 2022-11-24:
I can't reproduce the issue with the data I have... Do you have further samples I could utilize to try and reproduce?

eswnl, 2022-11-25:
Just sent the file.

Aang23, 2022-11-25:
Should be fixed!

eswnl, 2022-11-26:
I just downloaded newest build and still get the same error message.

Aang23, 2022-11-26:
Well, interesting. I guess I'll check on Windows.

Aang23, 2022-11-27:
Should be good now.

closed 2022-11-27 by Aang23

eswnl, 2022-11-27:
Hmm.
[image]

Aang23, 2022-11-27:
Hm... So it looks there's a separate issue entirely on Windows. Let me see...

Aang23, 2022-11-27:
Ok, figured it out... Another weird windows thing.

eswnl, 2022-11-28:
Working now. I got the "empty png" message. It also says there is 11000x11000 file but it did not output the image for it.
[image]
I'll keep playing.