SatDump/SatDump #195: Suggestion, signal quality and signal strength on geostationary satellites

nmoon88, 2022-11-24:
The current display of Viterbii is one I have NEVER seen before and this is coming from someone with seven years experience in FTA satellite television dealing with DVB-S/S2 it and the SNR measurements go so fast that there is no hope of accurately aligning a dish to pickup geostationary satellites with linear polarization. The vast majority of satellite receivers display signal quality and signal strength in the form of a bar chart which goes from 0% to 100% with most receivers showing both at the same time and the color of the bar changing from red to yellow to green as the signal strength and quality increase. This helps when dialing in the three adjustments required for linear polarized satellites which are azimuth or the heading at which the dish is pointing, elevation or how many degrees above the horizon it is pointing and skew or how many degrees the feedhorn or LNB is rotated. Optimizing all three is required for reliable reception and the current display makes this impossible so it makes aiming at a target 22,300 miles above the equator more difficult than it already is. Implementing a feature like this would be very helpful in fine tuning the alignment of a dish or the more common WIFI grid antenna. Otherwise this is an excellent piece of software and everything else I have to say about it despite the Windows image reception bug which I have gotten around by compiling and running it in a Linux VM is positive.

Aang23, 2022-11-27:
Hello!
Actually, what SatDump displays is "lower-level". Instead of showing a computed "signal strength", the raw values out of the demodulator (SNR), error correction (eg, Viterbi and RS for DVB-S, LDPC and BCH for DVB-S2) are shown. It's not as intuitive, but the idea is pretty much "get the SNR as high as possible, error rates as low as possible". 

Though, usually it's a better idea to look at the constellation diagram, and work off that while in the field. And use the SNR readings for more long-term comparisons. Everything such as Viterbi, etc is more of an indication of how good your lock is than signal level.

I'd perhaps agree slowing down the averages to make them more readable could a good idea, but computing a generic "signal strength" meter seems superfluous compared to other indicators - which already pretty much act this way.

Otherwise, I see you've also experienced the rice decompression error? What windows build / system was this on? I've been having a really hard time figuring it out and narrowing it down to a specific system configuration.

nmoon88, 2022-11-27:
The issue is occurring with a brand name PC, a Dell Optiplex 9010  which has an Intel Core i7 3770 at 3.4 GHz with 16 GB of RAM, a 1 TB hard drive and an nVidia GT1030 GPU which has been added on. As far as which Windows 10 build it is 19041.vb_release, 191206-1406 and I have had the same Rice Decompression Error on another Windows 10 computer this time a Dell Stream 11 G4 EE which is completely stock running Windows 10 Pro 22H2 with the Windows Feature Experience Pack 120.2212.4180.0 installed.  the Dell Optiplex had SDRPlay installed on it but removing that as suggested in the related thread did not do anything and the crash still remains. The HP Stream 11 has never had SDRPlay installed on it and yet the problem still occurs. I found the log file for the software and can include it if that would help for both computers running both builds of Windows.

Aang23, 2022-11-27:
Thanks for the information! I will need to try with those specific Windows builds to see if I can reproduce the issue.
As for the SDRPlay issue, that's pretty weird. The API should just fail to initialize as it's not been installed and not be loaded at all. I wouldn't mind the log file to check!

closed 2022-12-24 by Aang23