SatDump/SatDump #197: Progress bar/splash screen when updating TLEs on startup

RAD750, 2022-11-25:
**What would you want to be added to satdump?**
A progress bar or splash screen when updating TLEs on startup.

**Describe the feature you'd like to have**
A simple progress bar or splash screen that tells me it's updating TLEs, so I don't think the program has crashed on startup, click a second time and then two copies of SatDump open :D

**Describe alternatives you've considered**
Modifying the .desktop file so it opens it in a terminal instead, to see the debug info

Just a quick suggestion, I won't be offended if you just say "no". Thanks for the great software :heart:

Aang23, 2022-11-25:
Hello!
Honestly, you're making a very good point. I myself actually mainly use the CLI (even to start UI processing - I just use commands more) so nothing I really ever noticed... But the app just starting in the background with 0 user feedback is pretty bad.

Actually, I'd perhaps go further and suggest doing all initialization after creating an initial window.
Eg, creating a very basic window only showing perhaps a logo and logs of SatDumo starting up in a terminal-like fashion, then switching to the usual UI once said init finishes.

While not perfect, it'd let the user reading the logs have an idea of what's going on or just see something *is* going on.

Thanks for pointing this one out!

RAD750, 2022-11-27:
Also the same would apply when clicking on "Update TLEs" in the settings. Now, it just causes the app to freeze totally (Gnome and Android both say "App not responding") until the TLEs are update, with no feedback at all :)

Aang23, 2022-11-27:
@RAD750 just fixed this. Won't hand anymore.
I'll take a look at a splash screen a bit later. That will be some more work.

closed 2023-01-23 by Aang23