SatDump/SatDump #199: Show FFT when calling satdump record or satdump live from the command line

SV1CAL, 2022-11-26:
**What would you want to be added to satdump?**
The ability to see the spectrum while recording or live processing from the command line.
This will enable minor on-the-fly antenna adjustments .

**Describe the feature you'd like to have**
There could be an additional argument in the command line that enables or not the FFT spectrum like the checkbox in the GUI.

**Describe alternatives you've considered**
If enabling the spectrum in non-gui mode is tricky, some power reading that is regularly printed out  to allow antenna adjustments would also be good enough.

**Additional context**
I am calling SatDump from the command line in my satellite tracking/processing application and while on L band tracking accuracy 
is not critical for my 1.4m dish, on X band it is very useful to be able to make small adjustments. 

Thanks a lot !!!

Regards,
Michael SV1CAL

ZbychuButItWasTaken, 2022-11-26:
Having the FFT (or any part of the UI) in the CLI mode is impossible. 
However, you can launch satdump with an UI from the command line (by calling ./satdump-ui with the same arguments you'd use for ./satdump).

There is also information about the state of decoding printed to the console while live processing (every 10s).
[image]

SV1CAL, 2022-11-27:
Hello,
Thanks for your reply!
Yes using the UI would we be perfectly fine if all arguments could be passed from command line as in the CLI version like sdr, frequency, sampling rate etc
 
I tried replacing satdump.exe with satdump-ui.exe keeping the same arguments but neither live nor record are accepted : (C) Pipeline live does not exist " .
Same for record. So this didn't work for me.

In my application using the CLI version , satellites are tracked / recorded automatically without user intervention. A signal indication while recording would be helpful though on X band .

Anyway , I am still looking into this. I may be missing something of course .

Regards,
Michael

SV1CAL, 2022-12-06:
> Ah, somehow missing this one. Sorry @SV1CAL!
> 
> Well, as Zbychu said this would sort of defeat the current purpose of the CLI. Making it able to also spawn a FFT window or such would make it depend on what it is meant not to. As for satdump-ui, things are implemented drastically differently for live-processing and recording being meant to be UI-based... So while supporting the same syntax is most definitely possible, it'd however be pretty cumbersome to get working right.
> 
> When you are live-processing (though, likely not on X-Band), a SNR readout is available in the stats system. See https://www.satdump.org/posts/basic-usage/#normal-use. It will provide you the live readings over http in JSON format for the various modules running.
> 
> Though, this of course doesn't work for the recorder where no demodulation is performed. There, perhaps a cut-down FFT stream could optionally be served (running the same code as the UI, but only streaming FFT buffers out), but I'm not 100% sure that's worth it either. Though granted could be useful... Not sure what protocol I'd use though. HTTP may be a bit too awful for the purpose.
> 
> Another suggestion I'd have would be utilizing SatDump's C++ API perhaps? It is quite some more work but considering what you are doing would give the ability to this sort of stuff.
> 
> @sergsero well, it doesn't actually require many changes at all, more in the code starting the UI side to handle this. If it's for IQ only though, then sdrpp / spyserver are the current options.

Hello Alan,
Thanks again for looking into this.
I definitely don't want you to spend time on this. 
Indeed for live processing the SNR indication is good enough.
During recording, what about just printing out the total signal power at the ADC next to the data size written to disk  ? That probably would be good enough for on-the-fly antenna adjustments.
I'll have to look at the API. That sounds interesting !

Thanks again, 
Michael

Aang23, 2023-01-05:
@SV1CAL this has been implemented. See https://github.com/altillimity/SatDump/issues/227.

closed 2023-01-05 by Aang23