SatDump/SatDump #200: When Lime SDR is used in command line mode, its gain arguments don't seem to be functional

SV1CAL, 2022-11-26:
**Description of the issue**
I may be missing something , but LIME SDR gain doesn't seem to change  from the command line using the arguments --lna_gain , --tia_gain , --pga_gain.

These arguments above are described here:
https://github.com/altillimity/SatDump/blob/master/docs/SDR-Options.md

Instead **--gain**  seems to work OK and sets the global gain ( as can be done with the vertical slider in HDSDR).

**Hardware (SDR/PC/OS)**
LimeSDR USB / Windows
**Version (Eg, 1.0.0, CI Build #171)**
1.0.1

Thanks !!!
Michael SV1CAL

Aang23, 2022-11-26:
Hi!
Well that's normal, the right argument is --gain now, but I didn't update the documentation yet apparently.

Aang23, 2022-11-26:
Edited the documentation.

closed 2022-11-26 by Aang23

SV1CAL, 2022-11-26:
Thanks a lot  !
Regards ,
Michael