SatDump/SatDump #201: Metop demod failure with message "expected >"

RAD750, 2022-11-27:
**Description of the issue**

When decoding a 4 MHz baseband of a Metop B pass, SatDump will fail with a cryptic error message "expected >".
No other message is shown and there appears to be nothing useful in the log either.
The only thing that gets produced is a single admin message and a cadu file (that does open in LeanHRPT and shows a huge missing chunk in the AVHRR and other instruments) .

The signal of the pass was strong with a snr of about 15 dB, the only unusual thing is that at a certain point the BER goes up to about 0.22, then lowers (Reed Solomon all red, but it doesn't desync).

**Hardware (SDR/PC/OS)**
Thinkpad X260 (Intel i5-6200u, Intel HD Graphics 520); PlutoSDR
Pop!_OS 22.04

**Version (Eg, 1.0.0, CI Build #171)**
1.0.1 built yesterday at 23.30

**Screenshots**

Cryptic error message

[image]

The moment when Reed Solomon goes all red

[image]

**CADUs and other data useful for debugging**

The log

[satdump.txt](https://github.com/altillimity/SatDump/files/10097723/satdump.txt)

The baseband (WARNING large file ~9GB, maybe use wget)

[baseband](http://proxima.a-centauri.com/stuff/baseband_1701300000Hz_21-43-04_26-11-2022.wav)

The output produced

[output](http://proxima.a-centauri.com/stuff/Metop_fail.zip)

Thank you again and sorry for finding yet another bug :smile: :heart:

ZbychuButItWasTaken, 2022-11-27:
Could you upload the logs anyway? Helps to narrow down where the bug happpens.

RAD750, 2022-11-27:
I have, I had to rename the extension to .txt or GitHub wouldn't let me upload the satdump.logs file

Aang23, 2022-11-27:
This sounds horribly similar to what I fixed yesterday. Are you sure you were using the latest commit and not 1.0.1?

Aang23, 2022-11-27:
Just processed your baseband with the current latest commit - processed fine. Mind trying with the latest CI build / commit as well?

There are some sampledrops though, just so you know.

RAD750, 2022-11-27:
I did git pull and recompile explicitly because I saw your message, but maybe I need to start from scratch. I'll recompile again and close the issue if it's still fixed (should be)

Aang23, 2022-11-27:
Are you using the PlutoSDR over USB? I've noticed the sampledrops happen a lot more over IP. Though not sure SDR++ supports USB, you might have to use SatDump's recorder for this (so, the option NOT called PlutoSDR IP).

RAD750, 2022-11-27:
Yep that is fixed over my end too. Yes I am using the pluto via USB, but I need to input an IP address. I'll see what can be done with Ryzerth then. Thanks a lot :heart:

closed 2022-11-27 by RAD750

Aang23, 2022-11-27:
Yes - but you are actually using IP-Over-USB. The Pluto emulates an USB-Ethernet adapter in a way. That's the issue, the actually throughput is lower than raw USB.
Perhaps try in SatDump to see if direct USB helps!

You're welcome!