SatDump/SatDump #204: Volk redefinition build errors on Ubuntu 22.04.1

rtlsdrblog, 2022-11-28:
**Description of the issue**

Cannot build satdump on Ubuntu 22.04.1. After running `make` I get the error:

```
[ 11%] Building CXX object src-core/CMakeFiles/satdump_core.dir/common/dsp/moving_average.cpp.o
/home/carl/satdump/src-core/common/dsp/moving_average.cpp:6:6: error: redefinition of 'volk_32fc_x2_add_32fc' as different kind of symbol
void volk_32fc_x2_add_32fc(lv_32fc_t *cVector,
     ^
/usr/include/volk/volk.h:1667:36: note: previous definition is here
extern VOLK_API p_32fc_x2_add_32fc volk_32fc_x2_add_32fc;
                                   ^
1 error generated.
make[2]: *** [src-core/CMakeFiles/satdump_core.dir/build.make:958: src-core/CMakeFiles/satdump_core.dir/common/dsp/moving_average.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:770: src-core/CMakeFiles/satdump_core.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
```

Suspecting a volk version issue I have tried completely cleaning Volk out and reinstalled via apt-get install libvok2-dev and also tried to build Volk from source. But to no avail, it always ends up with this error.

**Hardware (SDR/PC/OS)**

Ubuntu 22.04.1
i7-11800H

Aang23, 2022-11-28:
Thanks for reporting! Since it's old code I do not need anymore anyway, I just deleted it and things should work. I will test in docker before closing.

Aang23, 2022-11-28:
Passed the test in a docker 22.04 container.
[image]

closed 2022-11-28 by Aang23