SatDump/SatDump #206: Add NOAA APT support

lpmrfentazis, 2022-11-30:
It would be great if satdump learned how to recive and decode NOAA APT. Now this usually happens through a bundle of Orbitron + sdrsharp + virtual audio cable + WXToImg, which works very stably. Satdump with the support of this technology would help a lot!

Aang23, 2022-11-30:
Indeed. APT support has been requested quite a bit. It's been on the long-term todo for a while, but never a priority either... Though I think I'll take a bit of time to get that started soon enough - but won't give an ETA.

K4KDR, 2022-11-30:
Many thanks for a least considering APT!

-Scott,  K4KDR

Aang23, 2022-11-30:
No problem!
It wasn't really something I initially expected to have requests about honestly. I'm a bit surprised by how many people actually want it.

Aang23, 2022-12-01:
Improved the sync. Only need to parse wedges now.
[image]

K4KDR, 2022-12-01:
Great!!  Can't wait to try tonight's NOAA passes.

Many thanks!

Aang23, 2022-12-01:
As a note :
For now, it will work with basebands (like all other signals). It can also work with 16-bits WAV audio files often used in the APT decoding scene, but I didn't add the menu option for audio samplerate yet.

Hopefully it works OK for you!

Aang23, 2022-12-01:
I did a bit more cleanup & changes. @K4KDR maybe do update in the meantime!

[image]
[image]
[image]

K4KDR, 2022-12-01:
Will be sure to get the latest before tonight's passes.

Many thanks!!

lpmrfentazis, 2022-12-02:
Wonderful! One of these days I will put the station on permanent duty for testing

closed 2022-12-02 by lpmrfentazis