SatDump/SatDump #207: Meteor-M2 projection doesn't work (again)

RAD750, 2022-12-03:
**Description of the issue**
Follow up from #166  because I couldn't get yesterday evening's METEOR-M N2 pass to project, or to have the country outline (which is really useful on IR). Checkbox is disabled, it doesn't generate the low resolution projection.

Didn't try LRPT as I had just arrived home, so just had the time to grab the necessary for HRPT :joy: 

**Hardware (SDR/PC/OS)**
Thinkpad X260, Pop!_OS 22.04

**Version (Eg, 1.0.0, CI Build #171)**
1.0.1 compiled 30th of November

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
[baseband here](https://proxima.a-centauri.com/stuff/baseband_1700000000Hz_18-17-19_02-12-2022.wav)

RAD750, 2022-12-03:
on LRPT it does work (couldn't test with HRPT due to heavy rainfall)
![rgb_msu_mr_5_Equalized_projected](https://user-images.githubusercontent.com/12469744/205453687-528e8878-1a4b-
462f-a502-96eb3ec48dc3.png)

Aang23, 2022-12-03:
Well, this is weird. I did not touch literally anything and as of recently things still worked fine. Will test with your file anyway.

Aang23, 2022-12-03:
So I ran it through. I do get projections just fine (using the current latest commit)... Which specific build were you using?

RAD750, 2022-12-04:
Okay, I tested again with 1.0.1 and now the checkbox was clickable but nothing got generated (nothing in the log either)
Updated to 1.0.3-54070639 and now it works, although it generates in the wrong location (But I recall you said it's normal as the day has changed)
[image]

Aang23, 2022-12-04:
Ok, so it's indeed an old one that has been fixed a while ago. Great.

This reminds me I need to implement a manual UTC day input for those issues.

RAD750, 2022-12-04:
Confirmed working on this evening's Princess (Meteor-M N2) pass

[image]

Aang23, 2022-12-04:
Nice. I see you also added a little heatmap for yourself!

I guess I can close this then?

RAD750, 2022-12-04:
Yep, we can close
As for the heatmap it's just a python script with OpenCV (I don't even know how to start with C++) but it works rather well :)

closed 2022-12-04 by RAD750