SatDump/SatDump #208: The string "(E) Could not set thread priority!" for goes_gvar in the log

sergsero, 2022-12-05:
The commit #8d2f693 was making changes "Thread priority fixes". Unfortunately, for the EWS-G1 satellite (formerly GOES-13), the red string "(E) Could not set thread priority!" is saved in the log (release 1.0.2).
[image]

**What would you want to be added to satdump?**
Please, for Server/Client Network streaming add geo pipelines work with the FengYun-2 S-VISSR and GOES GVAR satellites.

Thanks for the great software!

Aang23, 2022-12-05:
@sergsero I've just added Server/Client streaming for both. As for the priority error, I will debug a bit later.

sergsero, 2022-12-06:
Thank you so much for the quick solution!

Aang23, 2022-12-08:
Should be fixed.