SatDump/SatDump #216: sample test files in the repo

rabarar, 2022-12-13:
would be great if there were a few sample files included in the repo to verify the operations after building from source

Aang23, 2022-12-13:
Perhaps it would be a good idea to have some repository where a few samples could be downloaded. I'd rather host that on satdump.org than Github though, simply due to the sheer size of the files.

rabarar, 2022-12-14:
That makes more sense… agree

radio-satellites, 2022-12-21:
@Aang23 perhaps also a frequency table? That would be helpful as well :)

agausmann, 2023-09-24:
It would be nice to have sample datasets to perform a comprehensive test of the decoders.

As a package maintainer and user on both x86_64 and aarch64, I want to test it on as many platforms/architectures as possible, especially because of the platform-specific accelerations that SatDump has.

Aang23, 2023-09-24:
You can find some here : https://www.satdump.org/files/