SatDump/SatDump #217: FY3D demod and deframes issue

SV1CAL, 2022-12-17:
Hello !

I have a FY3D/45 MSPS/HDSDR/Lime SDR recording where version 1.0.3 demod can't get locked while 0.0.3x does.
I tweaked a bit pll_alpha from originally 0.006 to 0.0189 to match 0.0.3x's value and then it got immediately locked.
SNR is in the about 10-12 dB .
Still, after updating pll_alpha even though the constellation looks nice and stable , the deframer never gets green . 
On the same setup other sats seem to be ok like Terra, FY3E , Aqua etc . (In some cases QPSK demod benefits from pll_bw around 0.0189 though) but the deframer worked on these sats.
I am uploading a wav sample file to my google drive..it takes some time but will post the link here later.

Thanks,
Michael

Aang23, 2022-12-17:
That's very weird, not something anyone else ever experienced as far as I know. (neither did I personally)

I will need to check the baseband out...

Changing the pll_bw on some setups can be normal, it's likely due to a frequency offset in your setup. Providing a value that always works is not trivial! I lowered it from 189 as that did lower low-SNR performances a bit. Perhaps try with 0.008 I've put now.

SV1CAL, 2022-12-17:
Wow that was fast ! :-)
Let me try 0.008. 
Any hint about the deframer ?
Thanks,
Michael

Aang23, 2022-12-17:
The deframer makes the least sense. Should never happen if viterbi locks... So I will need to see.

SV1CAL, 2022-12-17:
It still getting uploaded but it's very slow . Maybe Saturday afternoon isn't the best timing :-). It will be done in an hour.
Still 0.008 doesn't work but 0.0189 does  , see below:

[image]

[image]

SV1CAL, 2022-12-17:
Here is the link to the FY3D / 45 MSPS / Lime HDSDR recording :

https://drive.google.com/file/d/1uSkM1dpgwIfETmlTJWSVlNwNnB5q3GaQ/view?usp=share_link

Aang23, 2022-12-17:
Ok so first of all, the issue with the costas is because you have a big DC carrier : Enabling DC Blocking fixed it.

I can also see the deframer is trying to synchronize, but it appears to be dropping/distorded in a weird way. Perhaps a gain issue? Pretty hard to tell just like that, just the recording does seem a bit weird at first glance.

[image]
[image]

SV1CAL, 2022-12-17:
Thanks for looking into this!
Will try recording again , adjusting gain a bit , it may have been a bit too high .

Regards,
Michael

Aang23, 2022-12-17:
You're welcome! Also perhaps - for this kind of issue - you could check out the Matrix room listed in the Readme?

SV1CAL, 2022-12-17:
OK sure , will check it out !
Thanks !

closed 2022-12-17 by SV1CAL