SatDump/SatDump #219: The spectrum display broke in 1.0.3

lpmrfentazis, 2022-12-19:
Just now I got around to testing live APT processing, but now I noticed that the spectrum and waterfall display does not work for me, the situation will be saved on different computers and sdr. As I understand it, the survey itself works, see screenshot 3-4.

AIRSPY SDR/ RTLSDR
Windows 10/11

1.0.3

There was no problem on 1.0.2.
[image]
[image]

For the test, I tried to run APT processing and spam with a walkie-talkie at the desired frequency, there is a reaction.

[image]
[image]

Aang23, 2022-12-19:
Well, first of all I would suggest you drag the "FFT Min" slider under the "FFT" Tab down. The default seems to high. Might as well lower it as it confused a few people.

lpmrfentazis, 2022-12-19:
> Ну, во-первых, я бы посоветовал вам перетащить ползунок «FFT Min» на вкладке «FFT» вниз. Значение по умолчанию кажется высоким. С тем же успехом можно было бы и понизить, так как это смутило нескольких человек.

Yes, but it doesn't seem to cancel the problem?

Aang23, 2022-12-19:
What do you see when you do that? Set it to be on the left, down to -120 or such?

lpmrfentazis, 2022-12-19:
Oh, it really seems to work.. And what could be the reason for such a difference in the displayed level with the same settings on different versions?
[image]
It turned out to be quite unexpected for me.

Aang23, 2022-12-19:
Play around with the FFT Max / Min settings, and perhaps the FFT Rate and Avg as well. Everything was modified so you will want to tune it etc.

I'll also take some feedback if you think some things don't work as they should!

lpmrfentazis, 2022-12-19:
> Play around with the FFT Max / Min settings, and perhaps the FFT Rate and Avg as well. Everything was modified so you will want to tune it etc.
> 
> I'll also take some feedback if you think some things don't work as they should!

In general, this does not apply to this issue, but when I build satdump on Ubuntu sources or install the apk on Android, SDRs are not detected, what could this be related to? That is, when connected, they simply do not appear in the list.

closed 2022-12-21 by lpmrfentazis