SatDump/SatDump #227: Add the current level in Db to the status server

lpmrfentazis, 2023-01-02:
**What would you want to be added to satdump?**
Hi! If it's not too difficult, I would like the status server to also transmit the current level. That is, in other words, not only the signal level above the noise, but also the noise level itself in Db.  This would allow me to integrate SatDump into an already existing system and get rid of the terrible SoapySDR scripts. The I level allows you to monitor the status of the signal line, LNA, and so on.[image]

[image]

[image]

Aang23, 2023-01-05:
Would a solution similar to #199 where I would include a FFT in the stats (optionally) work for you?

lpmrfentazis, 2023-01-05:
> Would a solution similar to #199 where I would include a FFT in the stats (optionally) work for you?

To be honest, I read this thread again, but in the end I didn't fully understand what the final version was planned.

Is it about streaming part of the spectrum ? In what form will the fft be transmitted?
For the most part, I am only interested in one number showing the noise level, in addition to the signal level.

Aang23, 2023-01-05:
You can now get the FFT values in stats if you enable the option. This works in both live and record mode.

--fft_enable // To enable it
--fft_size // Set FFT size, default 512
--fft_rate // Set FFT rate, default 30 per second
--fft_avg / Set FFT Avg rate

[image]

closed 2023-01-05 by Aang23