SatDump/SatDump #228: GK-2A lrit images blank

eswnl, 2023-01-03:
**Description of the issue**
GK-2A LRIT images are blank after decoding.

**Hardware (SDR/PC/OS)**
Airspy R1

**Version (Eg, 1.0.0, CI Build #171)**
v.104

**Logs after the cras (satdump.logs)**
[01/03/23 - 16:31:20] (I) Progress -nan(ind)%, SNR : 13.494963dB, Peak SNR: 14.311906dB
[01/03/23 - 16:31:20] (I) Progress -nan(ind)%
[01/03/23 - 16:31:20] (I) Progress -nan(ind)%, Viterbi : SYNCED BER : 0.010986, Deframer : SYNCED
[01/03/23 - 16:31:30] (I) Progress -nan(ind)%, SNR : 13.618789dB, Peak SNR: 14.311906dB
[01/03/23 - 16:31:30] (I) Progress -nan(ind)%, Viterbi : SYNCED BER : 0.009766, Deframer : SYNCED
[01/03/23 - 16:31:30] (I) Progress -nan(ind)%
[01/03/23 - 16:31:32] (I) Decrypting....
[01/03/23 - 16:31:32] (I) Decompressing JPEG...
[01/03/23 - 16:31:33] (I) New LRIT file : IMG_FD_044_IR105_20230103_073006_07.lrit
[01/03/23 - 16:31:33] (D) This is image data. Size 2200x220
[01/03/23 - 16:31:33] (D) JPEG Compression is used, decompressing...
[01/03/23 - 16:31:33] (D) This is encrypted!
[01/03/23 - 16:31:40] (I) Progress -nan(ind)%, SNR : 13.881208dB, Peak SNR: 14.311906dB
[01/03/23 - 16:31:40] (I) Progress -nan(ind)%, Viterbi : SYNCED BER : 0.004883, Deframer : SYNCED
[01/03/23 - 16:31:40] (I) Progress -nan(ind)%
[01/03/23 - 16:31:48] (I) Decrypting....
[01/03/23 - 16:31:48] (I) Decompressing JPEG...
[01/03/23 - 16:31:49] (I) New LRIT file : IMG_FD_044_IR105_20230103_073006_08.lrit
[01/03/23 - 16:31:49] (D) This is image data. Size 2200x220
[01/03/23 - 16:31:49] (D) JPEG Compression is used, decompressing...
[01/03/23 - 16:31:49] (D) This is encrypted!
[01/03/23 - 16:31:50] (I) Progress -nan(ind)%, SNR : 13.737448dB, Peak SNR: 14.311906dB
[01/03/23 - 16:31:50] (I) Progress -nan(ind)%, Viterbi : SYNCED BER : 0.006104, Deframer : SYNCED
[01/03/23 - 16:31:50] (I) Progress -nan(ind)%
[01/03/23 - 16:32:00] (I) Progress -nan(ind)%, SNR : 13.615156dB, Peak SNR: 14.311906dB
[01/03/23 - 16:32:00] (I) Progress -nan(ind)%, Viterbi : SYNCED BER : 0.008545, Deframer : SYNCED
[01/03/23 - 16:32:00] (I) Progress -nan(ind)%
[01/03/23 - 16:32:03] (I) Decrypting....
[01/03/23 - 16:32:03] (I) Decompressing JPEG...
[01/03/23 - 16:32:04] (I) New LRIT file : IMG_FD_044_IR105_20230103_073006_09.lrit
[01/03/23 - 16:32:04] (D) This is image data. Size 2200x220
[01/03/23 - 16:32:04] (D) JPEG Compression is used, decompressing...
[01/03/23 - 16:32:04] (D) This is encrypted!
[01/03/23 - 16:32:10] (I) Progress -nan(ind)%, SNR : 13.766565dB, Peak SNR: 14.311906dB
[01/03/23 - 16:32:10] (I) Progress -nan(ind)%, Viterbi : SYNCED BER : 0.008545, Deframer : SYNCED
[01/03/23 - 16:32:10] (I) Progress -nan(ind)%
[01/03/23 - 16:32:14] (I) Decrypting....
[01/03/23 - 16:32:14] (I) Decompressing JPEG...
[01/03/23 - 16:32:15] (I) New LRIT file : IMG_FD_044_IR105_20230103_073006_10.lrit
[01/03/23 - 16:32:15] (D) This is image data. Size 2200x220
[01/03/23 - 16:32:15] (D) JPEG Compression is used, decompressing...
[01/03/23 - 16:32:15] (D) This is encrypted!
[01/03/23 - 16:32:20] (I) Progress -nan(ind)%, SNR : 13.677044dB, Peak SNR: 14.311906dB
[01/03/23 - 16:32:20] (I) Progress -nan(ind)%, Viterbi : SYNCED BER : 0.007324, Deframer : SYNCED
[01/03/23 - 16:32:20] (I) Progress -nan(ind)%
[01/03/23 - 16:32:20] (I) Decrypting....
[01/03/23 - 16:32:20] (I) Decompressing JPEG...
[01/03/23 - 16:32:20] (I) Writing image ./live_output/2023-01-03_07-27_gk2a_lrit_1692Mhz/IMAGES/IMG_FD_044_IR105_20230103_073006.png...
[01/03/23 - 16:32:30] (I) Progr

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
[image]

eswnl, 2023-01-06:
Sent one just now. Sampling rate is 312500.

simonphangan, 2023-01-10:
I also have the same problem, using an RTL-SDR OEM #0

simonphangan, 2023-01-11:
I am running Satdump on Windows 10.  The unecrypted images (fog, pressure etc) are all received/stored OK in the ADD folder.  The encrypted image slices 01 to 10 are all received with good S/N, but the stored/decrypted png image is completely black in colour - an example just received and stored on my server at:

http://onlinescienceteacher.com/IMG_FD_035_IR105_20230111_060006.png

eswnl, 2023-01-11:
I remember previous versions had it working properly, so it could just be an accidental deletion somewhere during updating.

simonphangan, 2023-01-11:
I have just tried versions 1.02 and 1.00.  Both have the same problem (black png).  In v. 1.03, there is no signal displayed in the S/N window at all, but this is shown in v. 1.02 and v 1.00.  The command line processing in v 1.00 and v 1.02 shows no errors - all image slices are received and the png image is written to file.  But it's black!

eswnl, 2023-01-13:
You can revert back to v0.0.37 in the meantime, if you want to still get images.
[image]

simonphangan, 2023-01-13:
Thanks! 

I tried this version and indeed image slices are displayed - see attached screenshot.  But there seems to be some form of lost of sync in the image.  Any ideas?  Thanks.

[image]

eswnl, 2023-01-13:
Have you tried using a different sampling rate?

simonphangan, 2023-01-13:
Yes, I tried 1.0M and 2.048M, but with the same result.  It's very strange.  With this version you decode the images OK? What sampling rate?  Thanks

eswnl, 2023-01-13:
Can't really remember but if it was airspy, probably 10, 6 or 3 MSps.

simonphangan, 2023-01-13:
OK, I solved it!  After extracting the encryptionkey bin file from the zip, you need to decrypt it by running the .exe program.  That gives you a bin.dec file.  Then you rename that file to a .bin file again and store that in the GK-2A directory. That allows correct decoding of the image!  Phew!  I was just trying to use the .bin file, and not the bin.dec file......

eswnl, 2023-01-13:
Yes, I remember now. It seems that SatDump does not decrypt the key message file, so this is something you need to do manually.
[image]

closed 2023-01-13 by eswnl

Aang23, 2023-01-14:
Ah, nice you all figured it out. I will probably just automatically download and setup the keys on first start (of gk-2a) instead I think, easy enough to do and would avoid this.