SatDump/SatDump #229: Problems when building satdump on macbook M1

lpmrfentazis, 2023-01-03:
Problems when building satdump on macbook M1

Macbook air M1
16 gb 
MacOS Ventura 13.0.1

1.0.3 last comit (0a3a946)
yarik@MacBook-Air-Yarik build % arch -arm64 make -j4

`[ 50%] Building CXX object src-core/CMakeFiles/satdump_core.dir/resources.cpp.o
clang: warning: argument unused during compilation: '-L/usr/local/opt/libomp/lib' [-Wunused-command-line-argument]

[ 50%] Building CXX object src-core/CMakeFiles/satdump_core.dir/satdump_vars.cpp.o
clang: warning: argument unused during compilation: '-L/usr/local/opt/libomp/lib' [-Wunused-command-line-argument]

[ 50%] Building CXX object src-core/CMakeFiles/satdump_core.dir/spdlog/async.cpp.o
clang: warning: argument unused during compilation: '-L/usr/local/opt/libomp/lib' [-Wunused-command-line-argument]

[ 50%] Building CXX object src-core/CMakeFiles/satdump_core.dir/spdlog/cfg.cpp.o
clang: warning: argument unused during compilation: '-L/usr/local/opt/libomp/lib' [-Wunused-command-line-argument]

[ 50%] Building CXX object src-core/CMakeFiles/satdump_core.dir/spdlog/color_sinks.cpp.o
clang: warning: argument unused during compilation: '-L/usr/local/opt/libomp/lib' [-Wunused-command-line-argument]
4 warnings generated.

[ 50%] Building CXX object src-core/CMakeFiles/satdump_core.dir/spdlog/file_sinks.cpp.o
clang: warning: argument unused during compilation: '-L/usr/local/opt/libomp/lib' [-Wunused-command-line-argument]

[ 51%] Building CXX object src-core/CMakeFiles/satdump_core.dir/spdlog/fmt.cpp.o
clang: warning: argument unused during compilation: '-L/usr/local/opt/libomp/lib' [-Wunused-command-line-argument]

[ 51%] Building CXX object src-core/CMakeFiles/satdump_core.dir/spdlog/spdlog.cpp.o
clang: warning: argument unused during compilation: '-L/usr/local/opt/libomp/lib' [-Wunused-command-line-argument]

[ 51%] Building CXX object src-core/CMakeFiles/satdump_core.dir/spdlog/stdout_sinks.cpp.o
clang: warning: argument unused during compilation: '-L/usr/local/opt/libomp/lib' [-Wunused-command-line-argument]

[ 51%] Linking CXX shared library ../libsatdump_core.dylib
ld: warning: directory not found for option '-L/usr/local/opt/libomp/lib'
ld: library not found for -lomp
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libsatdump_core.dylib] Error 1
make[1]: *** [src-core/CMakeFiles/satdump_core.dir/all] Error 2
make: *** [all] Error 2`

Which way should I dig ?

Aang23, 2023-01-03:
Did you `brew install libomp` ?

lpmrfentazis, 2023-01-03:
> Did you `brew install libomp` ?

That's the point
libomp 15.0.6 is already installed and up-to-date

It looks like there are some problems with cmake's unwillingness to look for dependencies in M1-specific directories.

Aang23, 2023-01-03:
Where is libomp installed on your end?
I do not have a M1 machine to test things with, so any differences with x86_64 weren't covered.

lpmrfentazis, 2023-01-03:
> Where is libomp installed on your end? I do not have a M1 machine to test things with, so any differences with x86_64 weren't covered.

/opt/homebrew/opt/libomp/lib

Aang23, 2023-01-03:
Interesting. Does if work if you edit the CMakeList to use this path instead?

lpmrfentazis, 2023-01-03:
> Interesting. Does if work if you edit the CMakeList to use this path instead?

You mean?
link_directories("/opt/homebrew/opt/libomp/lib")

Aang23, 2023-01-03:
Yes

lpmrfentazis, 2023-01-03:
> Yes

Looks like I'll have to do this for all dependencies. I'll try it now

ld: warning: directory not found for option '-L/usr/local/opt/libomp/lib'
ld: library not found for -lvolk
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libsatdump_core.dylib] Error 1
make[1]: *** [src-core/CMakeFiles/satdump_core.dir/all] Error 2

Aang23, 2023-01-03:
Well, that's interesting. So homebrew puts everything in a very different directory on M1?

lpmrfentazis, 2023-01-03:
> Well, that's interesting. So homebrew puts everything in a very different directory on M1?

It looks like it is. I am not an experienced Mac user myself, I work with a colleague's computer.

Also, it seems to me, there may be a lot of problems due to rosetta (x86_64 emulator), since when using commands like "brew install name" macos tries to use it. To use native versions, you need to add "arch -arm64" to the beginning of the commands.

`-- Build files have been written to: /Users/yarik/satdump/satdump/build
[  0%] Linking CXX shared library ../libsatdump_core.dylib
ld: warning: directory not found for option '-L/usr/local/opt/libomp/lib'
ld: warning: ignoring file /usr/local/lib/libfftw3f.3.dylib, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
Undefined symbols for architecture arm64:
  "_fftwf_destroy_plan", referenced from:
      dsp::FFTPanBlock::~FFTPanBlock() in fft_pan.cpp.o
      dsp::FFTPanBlock::destroy_fft() in fft_pan.cpp.o
      dsp::FFTPanBlock::set_fft_settings(int, unsigned long long, int) in fft_pan.cpp.o
      image::fft_forward(image::Image<unsigned short>&) in fft.cpp.o
      image::fft_inverse(image::Image<unsigned short>&) in fft.cpp.o
  "_fftwf_execute", referenced from:
      dsp::FFTPanBlock::work() in fft_pan.cpp.o
      image::fft_forward(image::Image<unsigned short>&) in fft.cpp.o
      image::fft_inverse(image::Image<unsigned short>&) in fft.cpp.o
  "_fftwf_free", referenced from:
      dsp::FFTPanBlock::~FFTPanBlock() in fft_pan.cpp.o
      dsp::FFTPanBlock::destroy_fft() in fft_pan.cpp.o
      dsp::FFTPanBlock::set_fft_settings(int, unsigned long long, int) in fft_pan.cpp.o
  "_fftwf_malloc", referenced from:
      dsp::FFTPanBlock::set_fft_settings(int, unsigned long long, int) in fft_pan.cpp.o
  "_fftwf_plan_dft_1d", referenced from:
      dsp::FFTPanBlock::set_fft_settings(int, unsigned long long, int) in fft_pan.cpp.o
  "_fftwf_plan_dft_c2r_2d", referenced from:
      image::fft_inverse(image::Image<unsigned short>&) in fft.cpp.o
  "_fftwf_plan_dft_r2c_2d", referenced from:
      image::fft_forward(image::Image<unsigned short>&) in fft.cpp.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libsatdump_core.dylib] Error 1
make[1]: *** [src-core/CMakeFiles/satdump_core.dir/all] Error 2
make: *** [all] Error 2`

I realized this too late, I'll try to reinstall the dependencies.

Aang23, 2023-01-03:
Ah, that could be part of the issue. Native arm64 works fine (apart from CMake issues clearly)

lpmrfentazis, 2023-01-03:
> Ah, that could be part of the issue. Native arm64 works fine (apart from CMake issues clearly)

Yes, it is. I managed to assemble SatDump without any problems on raspberry and on more specific Radxa with a Rock chip. I'm trying to reinstall dependencies

Aang23, 2023-01-03:
I think you will have to track it down and rm -rf it manually...

lpmrfentazis, 2023-01-03:
> I think you will have to track it down and rm -rf it manually...

After trying to build libfftw from the sources for ARM, according to the recommendations from here: https://stackoverflow.com/questions/68991284/build-fftw-for-apple-m1-and-the-older-intel-devices

But after "arch -arm64 make" I get this message, and it seems I do not know how to fix it:

/Library/Developer/CommandLineTools/usr/lib/clang/14.0.0/include/mmintrin.h:14:2: error: "This header is only meant to be used on x86 and x64 architecture"

#error "This header is only meant to be used on x86 and x64 architecture"
 ^
/Library/Developer/CommandLineTools/usr/lib/clang/14.0.0/include/mmintrin.h:54:12: error: invalid conversion between vector type '__m64' (vector of 1 'long long' value) and integer type 'int' of different size
    return (__m64)__builtin_ia32_vec_init_v2si(__i, 0);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

/Library/Developer/CommandLineTools/usr/lib/clang/14.0.0/include/mmintrin.h:133:12: error: invalid conversion between vector type '__m64' (vector of 1 'long long' value) and integer type 'int' of different size
    return (__m64)__builtin_ia32_packsswb((__v4hi)__m1, (__v4hi)__m2);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

/Library/Developer/CommandLineTools/usr/lib/clang/14.0.0/include/mmintrin.h:163:12: error: invalid conversion between vector type '__m64' (vector of 1 'long long' value) and integer type 'int' of different size
    return (__m64)__builtin_ia32_packssdw((__v2si)__m1, (__v2si)__m2);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

/Library/Developer/CommandLineTools/usr/lib/clang/14.0.0/include/mmintrin.h:193:12: error: invalid conversion between vector type '__m64' (vector of 1 'long long' value) and integer type 'int' of different size
    return (__m64)__builtin_ia32_packuswb((__v4hi)__m1, (__v4hi)__m2);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

/Library/Developer/CommandLineTools/usr/lib/clang/14.0.0/include/mmintrin.h:220:12: error: invalid conversion between vector type '__m64' (vector of 1 'long long' value) and integer type 'int' of different size
    return (__m64)__builtin_ia32_punpckhbw((__v8qi)__m1, (__v8qi)__m2);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

/Library/Developer/CommandLineTools/usr/lib/clang/14.0.0/include/mmintrin.h:243:12: error: invalid conversion between vector type '__m64' (vector of 1 'long long' value) and integer type 'int' of different size
    return (__m64)__builtin_ia32_punpckhwd((__v4hi)__m1, (__v4hi)__m2);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

/Library/Developer/CommandLineTools/usr/lib/clang/14.0.0/include/mmintrin.h:264:12: error: invalid conversion between vector type '__m64' (vector of 1 'long long' value) and integer type 'int' of different size
    return (__m64)__builtin_ia32_punpckhdq((__v2si)__m1, (__v2si)__m2);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

/Library/Developer/CommandLineTools/usr/lib/clang/14.0.0/include/mmintrin.h:291:12: error: invalid conversion between vector type '__m64' (vector of 1 'long long' value) and integer type 'int' of different size
    return (__m64)__builtin_ia32_punpcklbw((__v8qi)__m1, (__v8qi)__m2);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

/Library/Developer/CommandLineTools/usr/lib/clang/14.0.0/include/mmintrin.h:314:12: error: invalid conversion between vector type '__m64' (vector of 1 'long long' value) and integer type 'int' of different size
    return (__m64)__builtin_ia32_punpcklwd((__v4hi)__m1, (__v4hi)__m2);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

/Library/Developer/CommandLineTools/usr/lib/clang/14.0.0/include/mmintrin.h:335:12: error: invalid conversion between vector type '__m64' (vector of 1 'long long' value) and integer type 'int' of different size
    return (__m64)__builtin_ia32_punpckldq((__v2si)__m1, (__v2si)__m2);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

/Library/Developer/CommandLineTools/usr/lib/clang/14.0.0/include/mmintrin.h:356:12: error: invalid conversion between vector type '__m64' (vector of 1 'long long' value) and integer type 'int' of different size
    return (__m64)__builtin_ia32_paddb((__v8qi)__m1, (__v8qi)__m2);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

/Library/Developer/CommandLineTools/usr/lib/clang/14.0.0/include/mmintrin.h:377:12: error: invalid conversion between vector type '__m64' (vector of 1 'long long' value) and integer type 'int' of different size
    return (__m64)__builtin_ia32_paddw((__v4hi)__m1, (__v4hi)__m2);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

/Library/Developer/CommandLineTools/usr/lib/clang/14.0.0/include/mmintrin.h:398:12: error: invalid conversion between vector type '__m64' (vector of 1 'long long' value) and integer type 'int' of different size
    return (__m64)__builtin_ia32_paddd((__v2si)__m1, (__v2si)__m2);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

/Library/Developer/CommandLineTools/usr/lib/clang/14.0.0/include/mmintrin.h:420:12: error: invalid conversion between vector type '__m64' (vector of 1 'long long' value) and integer type 'int' of different size
    return (__m64)__builtin_ia32_paddsb((__v8qi)__m1, (__v8qi)__m2);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

[... cut ...] 2: error: invalid conversion between vector type '__m64' (vector of 1 'long long' value) and integer type 'int' of different size
    return (__m64)__builtin_ia32_paddsw((__v4hi)__m1, (__v4hi)__m2);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [CMakeFiles/fftw3f.dir/dft/simd/sse2/codlist.c.o] Error 1
make[1]: *** [CMakeFiles/fftw3f.dir/all] Error 2
make: *** [all] Error 2

Aang23, 2023-01-03:
Do not do
`cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_TESTS=false -DENABLE_FLOAT=true -DENABLE_THREADS=true -DENABLE_SSE=true -DENABLE_SSE2=true -DENABLE_AVX=true -DENABLE_AVX2=true`

But `cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_TESTS=false -DENABLE_FLOAT=true -DENABLE_THREADS=true -DENABLE_NEON=true`

lpmrfentazis, 2023-01-03:
> 

/Users/yarik/satdump/fftw-3.3.9/simd-support/x86-cpuid.h:137:16: error: invalid output constraint '+a' in asm
             : "+a" (*eax), "+r" (*ebx), "+c" (*ecx), "+d" (*edx));
               ^
/Users/yarik/satdump/fftw-3.3.9/simd-support/x86-cpuid.h:162:9: error: invalid output constraint '=a' in asm
             : "=a" (eax), "=c" (ecx), "=d" (edx)
               ^
/Users/yarik/satdump/fftw-3.3.9/simd-support/x86-cpuid.h:184:9: error: invalid output constraint '=a' in asm
             : "=a" (eax), "=c" (ecx), "=d" (edx)
               ^
/Users/yarik/satdump/fftw-3.3.9/simd-support/x86-cpuid.h:209:42: error: invalid output constraint '=a' in asm
     __asm__ (".byte 0x0f, 0x01, 0xd0" : "=a"(eax), "=d"(edx) : "c" (op));
     
     
In general, I understand why it doesn't work. ASM inserts are used there.. 
Yes, I'll try it now

closed 2023-01-14 by lpmrfentazis