SatDump/SatDump #236: Terminate called without an active exception. Aborted (core dumped).

sergsero, 2023-01-14:
Hello,

Background:
The Ubuntu 22.04. SatDump is started in the console via ./satdump-ui or via a shortcut. The mode of operation is 'live' (which, unfortunately, is now only called 'recording'). The data is received from a SDR remote via the 'sdrpp --server' in client-server mode.

After a receiving session with GEO satellites, when I need to close the program, I consistently press the 'Stop' keys in the sections: 'Processing', 'Device' and 'Disconnect'. After that I close the program by clicking on its "X" close button.

The program is closing. However, `"Terminate called without an active exception. Aborted (core dumped)"` is the message I see whenever I attempt to terminate the program in the console.
Or a pop-up system's window of Ubuntu's 'apport.service' that to report the program was crashed, if SatDump was starting via shortcut. 

Below is the example of problem messages after successfully receiving images from the FY-2H satellite.
[image]

Maybe the problem is related to threads?
Is it possible to make (an interrupt handler?) and call it by clicking on "X" close button? Or to use the new additional button to correct termination the program in GUI.

Best regards,
sergsero

Aang23, 2023-01-17:
Thanks for reporting, will have to try & reproduce, will see tomorrow.

Aang23, 2023-01-17:
Fixed!

closed 2023-01-17 by Aang23