SatDump/SatDump #237: Unpacking EOS-6 (OCEANSAT-3) data

lpmrfentazis, 2023-01-15:
**What would you want to be added to satdump?**
For the sake of interest, I tried to take the recently launched OCEANSAT-3 and process it via satdump. Found that the result is only CADU. Do I understand correctly that the data has been decrypted, but new products are not being created based on them?

**Describe the feature you'd like to have** 
Is it possible to add something similar to what is produced for AHRPT?

**Additional context**
Maybe this will help

[https://www.youtube.com/watch?v=7M_RXPJmcwo](url)
[https://ioccg.org/sensors/OCEANSAT_2.pdf](url)
[https://ioccg.org/sensors/OCM-2.pdf](url)

Perhaps this can also be applied to OTB3 GAzelle, where Argos also stands.

lpmrfentazis, 2023-01-15:
And there's only ARGOS. Other devices are unknown how to unpack ?

Aang23, 2023-01-16:
Unfortunately, L-Band contains ARGOS-4 data exclusively, in an identical fashion to ANGELS, SARAL and Gazelle. No instrument data is downlinked on L-Band, and as of now data transmission in X-Band has only been detected in range of Svalbard.

lpmrfentazis, 2023-01-16:
Is the information about the x-band accurate? I can try to take a couple of flights from a station located in Moscow.

lpmrfentazis, 2023-01-16:
But then I need its x-band frequency. And information about what I should do with iq, in case of a positive result

Aang23, 2023-01-16:
@MeteoOleg checked it, and I did as well... Only dumps as of now. If you wish to try it : 
8275Mhz, there's a pipeline in SatDump.

lpmrfentazis, 2023-01-16:
> @MeteoOleg checked it, and I did as well... Only dumps as of now. If you wish to try it : 8275Mhz, there's a pipeline in SatDump.

Ok  I'll try

closed 2023-02-02 by ZbychuButItWasTaken