SatDump/SatDump #238: Add audio source to live decoding

cserver45, 2023-01-16:
**What would you want to be added to satdump?**
An audio source option in the live decoding menu

**Describe the feature you'd like to have**
I'd like to have it because I pipe audio to a few programs in sdr++ while decoding noaa satellites.

**Describe alternatives you've considered**
Possibly use sdr++ server, but I run everything from the same laptop.
Also I don't think sdr++ server supports multiple connections, but I haven't done much research into it.

**Additional context**
None

Aang23, 2023-01-17:
That's possible, though in this case the expected input would be audio interpreted as IQ baseband.

Aang23, 2023-04-20:
Added a little while ago.

closed 2023-04-20 by Aang23