SatDump/SatDump #241: Processing of partially received S-VISSR data from GEO Feng-Yun-2x in incomplete sessions

sergsero, 2023-01-25:
**Describe the feature you'd like to have**

Hello,

From FY-2x satellites the S-VISSR data is transmitted in two sessions during each hour. From 00 minutes to ~00:28 minutes there is a full Earth disk. It ends with receiving the signal: "Full disk end detected!". After that, data processing begins.

But from 00:30 minutes to the end of the hour, data is transmitted only about ~half of this time interval. Except for a few sessions around noon. But this is enough to take up to 50% of the Northern hemisphere.

But how to save them?

If I'm stopped live processing by pressing the 'Stop' key, the received data is not processed and is lost.
It also happens (the received part of the data for the interval of the first ~15 minutes is lost) if the processing is not stopped but wait for the start of a new full session from 00 minutes of the next hour. The early received data will be lost too.

Is it possible to implement processing of partially received S-VISSR data in the same way as GOES GVAR? In the GVAR case, the data is fully processed and images are created for any duration of their receiving and interruption by the 'Stop' key or with the start transmission of a new next session from the satellite?

Sincerely,
sergsero

p.s. Adding the "Decimation in recorder" function is amazing! Thank you so much!

Aang23, 2023-01-25:
Yes, it is. Would you be able to record S-VISSR frames for a while so I have some data for testing?

sergsero, 2023-01-25:
Could you please tell me in which format it is better to record IQ using SatDump's recorder: s8 or s16?
How long should the recording be?
Would it be better if the recording contains the moment of data transmission interruption and the transmitter switches to an empty carrier?

I will make such a record as soon as possible.

Aang23, 2023-01-25:
No, perhaps only save the output of the S-VISSR decoder (before images, the .svissr) frames. It will be a lot smaller so you can collect hours of data or so to have enough to confirm things do work.

sergsero, 2023-01-26:
I'm somewhat confused, since data reception from FY-2H has been going on for the 4th hour. Images are created in 2023-01-26_06-38_fengyun_svissr_1687Mhz/IMAGE, in nested folders with date and reception times.

However, the required fengyun_svissr.svissr file is not being created. Neither when do receiving in the CLI, nor with the GUI.
Should this file be created automatically, without additional settings?

Maybe I'm doing something wrong?

sergsero, 2023-01-27:
In the log at the start of the program there are lines about creating several files, including `fengyun_svissr.svissr` and `fengyun_svissr.soft`, with indicating of the path to them.
However, apparently these files are not being created.

[image]

Aang23, 2023-01-27:
Yes, for it to show up you will want to modify the pipeline :
```
 "live_cfg": {
            "pkt_size": 44356,
            "default": [
                [
                    1,
                    0
                ],
                [
                    2,
                    0
                ],
                [
                    3,
                    0
                ]
            ],
            "server": [
                [
                    1,
                    0
                ],
                [
                    2,
                    0
                ]
            ],
            "client": [
                [
                    3,
                    0
                ]
            ]
        },

```

to
```
 "live_cfg": {
            "pkt_size": 44356,
            "default": [
                [
                    1,
                    0
                ],
                [
                    2,
                    0
                ]
            ],
            "server": [
                [
                    1,
                    0
                ],
                [
                    2,
                    0
                ]
            ],
            "client": [
                [
                    3,
                    0
                ]
            ]
        },
```

in pipelines/FengYun-2.json.

sergsero, 2023-01-28:
I posted the file `cli_fengyun_svissr.svissr.tgz` with the recording: [https://drive.google.com/file/d/1LXQjHO_5ZGgPYCu5lOO_RB1xo7PXoHoP/view?usp=sharing](url)
The recording duration is 8 hours and 30 minutes.

I checked it by processing it in the program. But I found that:
1. Only the first and last (by recording time) sets of images are created.
2. The actual time of receiving images is lost. The file names indicate the post-processing time. All files are placed in one common directory, and not sorted into different ones.

Also, it seems that the decimation coefficient works correctly in the GUI version. But in the live CLI, if it is set as `--decimation n`, it is displayed in the log, but probably not processed. The 'Samplerate', 'Dec factor', and 'Final PS' entries in the GUI/CLI logs are different.

pluggerlockett, 2023-02-09:
I've been 'up' on FY-2G for a little under two weeks. From what I can see it does hourly transmissions starting at hh:00 and ending at hh:25. It also does 'out of band' transmissions 4 times a day at a 6 hour interval during the following timeslots (all UTC). I expect it does this to get the fully illuminated FD image, but I'm not sure on that.

- 05:30
- 11:30
- 17:30
- 23:30

If I can grab any recordings to help please let me know.

[image]

Aang23, 2023-02-09:
My idea to fix this as of now is to rely on the "rollback" data we also get in S-VISSR. 

While scanning, the imager scans north to south, transmitting the full image. However, once done, it goes back to north a lot faster in a mode that is indicated by a bit in the data. Hence, instead of relying on the line counter indicating the south part of the scan to detect it, I would instead detect the imager starting to back to north. This would delay saving a bit, but main method I see to fix this.

I will do the changes in a separate branch, to allow anyone who can to text it out before merging.

Aang23, 2023-02-09:
@pluggerlockett I wouldn't mind some .svissr from 2G as well.

Aang23, 2023-02-09:
In the end, it was pushed on master. @sergsero @pluggerlockett if you both could test I wouldn't mind.

closed 2023-02-09 by Aang23

pluggerlockett, 2023-02-10:
> 

I've got some on a machine at home. I'll upload it and share the link here later today.

pluggerlockett, 2023-02-10:
I'll also reinstall tonight off master and see if anything else falls out.

pluggerlockett, 2023-02-10:
Here's a two transmission .svissr from a couple weeks ago @Aang23. It includes the full disk transmission.
[https://www.swisstransfer.com/d/9639f4c3-a1f5-446f-9698-007a581a6082](url)

I've also upgraded off master and have my station running v1.1.0-82340387. I'll keep an eye out for anything new.

sergsero, 2023-02-10:
Hello,

The first results of my testing - commit 8234038:
1. Directories are actually created after 44 minutes in sessions with partial transmission of the northern hemisphere. And after 55 minutes, when some sessions of the second half hour were with the full disk of the Earth.

[image]

[image]

[image]

2. Meanwhile, when manually interrupted by the Ctrl-C command, the CLI program stops working without processing the received data and image files are not created. The received data  were lost.

[image]

Unlike the GOES-13's algorithm, in which, when interrupted by the Ctrl-C, already received data is processed. And only after that, the program says goodbye and closes.

[image]

I will continue testing and observe the results.

Sincerely,
sergsero

Aang23, 2023-02-10:
@sergsero just fixed saving on live exit.

sergsero, 2023-02-11:
@Aang23 thank you very much for fixed!
Now when I interrupted receiving by 'Control-C', commit `f1e7d5d` successfully started processing the data received up to that point. It's beautiful, thanks!

I also recorded the new 8-hour fengyun_svissr.svissr file:
https://drive.google.com/file/d/1MAv9zsQ2LFxwi5oFvR8ozFLRfQK_lDMg/view?usp=sharing 
and then processed it offline. As a result, I got 16 images sorted into 7 directories. Now it works!

Are there probably limitations of the .svissr format that do not allow storing data about the time of each session and related images in it? But this is not essential.

Sincerely, sergsero

Aang23, 2023-02-11:
The .svissr file is simply raw frames from the S-VISSR stream. It's demodulated, synchronized and derandomized frames. They don't carry any timestamp in the format, so it has to be obtained live and there is no easy work-around.