SatDump/SatDump #242: LimeSDR mini is not visible on ubuntu 22.04

lpmrfentazis, 2023-01-25:
**Description of the issue**

LimeSDR mini is not displayed in the Recorder menu in the list of sources. I kind of installed all the dependencies, and Lime is displayed in both LimeUtil and SoapySDRUtil, but SatDump does not see it.

On the same computer, but on windows 10, LimeSDR is detected and running. It seems that I have something wrong with dependencies...

**Hardware (SDR/PC/OS)**
LimeSDR Mini
i5 12500, gigabite h610m sh2, 32 gb
Linux KDE neon 5.26 (Ubuntu 22.04 with graphical shell)

**Version (Eg, 1.0.0, CI Build #171)**
Last commit in Master

**Logs after the cras (satdump.logs)**
---
LimeUtil --find
[LimeSDR Mini, media=USB 3.0, module=FT601, addr=24607:1027, serial=1D7513FF30FE47]

SoapySDRUtil --probe="driver=lime"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=lime
[INFO] Make connection: 'LimeSDR Mini [USB 3.0] 1D7513FF30FE47'
[INFO] Reference clock 40.00 MHz
[INFO] Device name: LimeSDR-Mini
[INFO] Reference: 40 MHz
[INFO] LMS7002M register cache: Disabled

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=FT601
  hardware=LimeSDR-Mini
  boardSerialNumber=0x1d7513ff30fe47
  firmwareVersion=6
  gatewareVersion=1.30
  hardwareVersion=3
  protocolVersion=1

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 1 Tx
  Timestamps: YES
  Sensors: clock_locked, lms7_temp
     * clock_locked (Clock Locked): true
        CGEN clock is locked, good VCO selection.
     * lms7_temp (LMS7 Temperature): 34.485706 C
        The temperature of the LMS7002M in degrees C.
  Registers: BBIC
  Other Settings:
     * SAVE_CONFIG - Save LMS settings to file
       [key=SAVE_CONFIG, type=string]
     * LOAD_CONFIG - Load LMS settings from file
       [key=LOAD_CONFIG, type=string]
     * OVERSAMPLING - oversampling ratio (0 - auto)
       [key=OVERSAMPLING, type=int, options=(0, 1, 2, 4, 8, 16, 32)]
  GPIOs: MAIN

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: NO
  Stream formats: CF32, CS12, CS16
  Native format: CS16 [full-scale=32767]
  Stream args:
     * Buffer Length - The buffer transfer size over the link.
       [key=bufferLength, units=samples, default=0, type=int]
     * Latency - Latency vs. performance
       [key=latency, default=0.5, type=float]
     * Link Format - The format of the samples over the link.
       [key=linkFormat, default=CS16, type=string, options=(CS16, CS12)]
     * Skip Calibration - Skip automatic activation calibration.
       [key=skipCal, default=false, type=bool]
     * align phase - Attempt to align phase of Rx channels.
       [key=alignPhase, default=false, type=bool]
  Antennas: NONE, LNAH, LNAL_NC, LNAW, Auto
  Corrections: DC removal, DC offset, IQ balance
  Full gain range: [-12, 61] dB
    TIA gain range: [0, 12] dB
    LNA gain range: [0, 30] dB
    PGA gain range: [-12, 19] dB
  Full freq range: [0, 3800] MHz
    RF freq range: [30, 3800] MHz
    BB freq range: [-7.68, 7.68] MHz
  Tune args:
     * LO Offset - Tune the LO with an offset and compensate with the baseband CORDIC.
       [key=OFFSET, units=Hz, default=0.0, type=float, range=[-7.68e+06, 7.68e+06]]
     * BB - Specify a specific value for this component or IGNORE to skip tuning it.
       [key=BB, units=Hz, default=DEFAULT, type=float, range=[-7.68e+06, 7.68e+06], options=(DEFAULT, IGNORE)]
  Sample rates: [0.1, 30.72] MSps
  Filter bandwidths: [1.4001, 130] MHz
  Sensors: lo_locked
     * lo_locked (LO Locked): true
        LO synthesizer is locked, good VCO selection.
  Other Settings:
     * TSP_CONST - Digital DC test signal level in LMS7002M TSP chain.
       [key=TSP_CONST, default=16383, type=int, range=[0, 32767]]
     * CALIBRATE -  DC/IQ calibration bandwidth
       [key=CALIBRATE, type=float, range=[2.5e+06, 1.2e+08]]
     * ENABLE_GFIR_LPF - LPF bandwidth (must be set after sample rate)
       [key=ENABLE_GFIR_LPF, type=float]
     * TSG_NCO - Enable NCO test signal
       [key=TSG_NCO, default=4, type=int, options=(-1, 4, 8)]

----------------------------------------------------
-- TX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: NO
  Stream formats: CF32, CS12, CS16
  Native format: CS16 [full-scale=32767]
  Stream args:
     * Buffer Length - The buffer transfer size over the link.
       [key=bufferLength, units=samples, default=0, type=int]
     * Latency - Latency vs. performance
       [key=latency, default=0.5, type=float]
     * Link Format - The format of the samples over the link.
       [key=linkFormat, default=CS16, type=string, options=(CS16, CS12)]
     * Skip Calibration - Skip automatic activation calibration.
       [key=skipCal, default=false, type=bool]
     * align phase - Attempt to align phase of Rx channels.
       [key=alignPhase, default=false, type=bool]
  Antennas: NONE, BAND1, BAND2, Auto
  Corrections: DC offset, IQ balance
  Full gain range: [-12, 64] dB
    PAD gain range: [0, 52] dB
    IAMP gain range: [-12, 12] dB
  Full freq range: [0, 3800] MHz
    RF freq range: [30, 3800] MHz
    BB freq range: [-7.68, 7.68] MHz
  Tune args:
     * LO Offset - Tune the LO with an offset and compensate with the baseband CORDIC.
       [key=OFFSET, units=Hz, default=0.0, type=float, range=[-7.68e+06, 7.68e+06]]
     * BB - Specify a specific value for this component or IGNORE to skip tuning it.
       [key=BB, units=Hz, default=DEFAULT, type=float, range=[-7.68e+06, 7.68e+06], options=(DEFAULT, IGNORE)]
  Sample rates: [0.1, 30.72] MSps
  Filter bandwidths: [5, 40], [50, 130] MHz
  Sensors: lo_locked
     * lo_locked (LO Locked): true
        LO synthesizer is locked, good VCO selection.
  Other Settings:
     * TSP_CONST - Digital DC test signal level in LMS7002M TSP chain.
       [key=TSP_CONST, default=16383, type=int, range=[0, 32767]]
     * CALIBRATE -  DC/IQ calibration bandwidth
       [key=CALIBRATE, type=float, range=[2.5e+06, 1.2e+08]]
     * ENABLE_GFIR_LPF - LPF bandwidth (must be set after sample rate)
       [key=ENABLE_GFIR_LPF, type=float]
     * TSG_NCO - Enable NCO test signal
       [key=TSG_NCO, default=4, type=int, options=(-1, 4, 8)]

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

Aang23, 2023-01-25:
What does cmake .. say?

Aang23, 2023-01-25:
What does LimeUtil --find say?

lpmrfentazis, 2023-01-25:
> LimeUtil --find

LimeUtil --find
  * [LimeSDR Mini, media=USB 3.0, module=FT601, addr=24607:1027, serial=1D7513FF30FE47]

Aang23, 2023-01-25:
That is really weird... And no errors in satdump loading Limesuite?

lpmrfentazis, 2023-01-25:
> That is really weird... And no errors in satdump loading Limesuite?

I don't understand what you mean. Among the list of plugins that satdump loads, I didn't notice anything related to LimeSDR, isn't that the point? I previously left a cmake log where he reported that he could not find the UUID lib, perhaps it went unnoticed due to formatting

[image]

Aang23, 2023-01-25:
SatDump uses LimeSuite directly - Hence the Building with LimeSDR support.

You should have a limesdr_sdr_support. Do you see that when running make?

lpmrfentazis, 2023-01-25:
[image]

It seems now he has completed LimeSDR_support, which for some reason was not built earlier.  It would be convenient if the dependencies to support working with SDR were also specified in the assembly instructions.

Aang23, 2023-01-25:
Yes... A good idea. Should work now?

lpmrfentazis, 2023-01-25:
Thank you for what you are doing!
[image]

closed 2023-01-25 by lpmrfentazis