SatDump/SatDump #246: The OpenCL with Intel's GPU. Unix epoch timestamp error in the system with the national date and time format.

sergsero, 2023-02-04:
**What would you want to be added to satdump?**

1.	In Ubuntu for OpenCL, if Intel's GPU (iGPU) is used, along with `ocl-icd-opencl-dev`, probably should additionally install the `intel-opencl-icd` package. The Intel OpenCL runtime will be exposing CPU as an OpenCL device to the application.

2.	If the system uses the national date and time format (possibly not in Latin characters) and UTC offset, the date and time of the receiving sessions in the GUI Viewer is incorrectly specified. And it’s also wrong in the Unix Epoch `"timestamp": 1675842675.8152092` field in the created `dataset.json`.
At the same time, in the name of the created 'Live processing directories', the date and time are correct and specified in UTC.

For example, in reality, these sessions were one after the other on 2023-02-04 from 02:53 to 04:02UTC.
[image]

My long-term dreams :)

3.	Can the possibility be added for users to create an additional file with individual locations that are not in the original `ne_10m_populated_places_simple.json`, based on their local preferences? And can it be in a less complex format than ArcGIS, for example, CSV: `NAME,LAT,LON`?

4.	Since it has become possible to receive narrow-band signals like STD-C and AERO, is it possible to implement a horizontal frequency scale (x-axis) in the spectrum (FFT) window?
And also make it more convenient to enter and/ or change each digit of the frequency, for example, by changing them with a mouse wheel? Maybe it will be like a similar spectrum window in sdr++? Or are there fundamental limitations here?

Best regards,
sergsero

ZbychuButItWasTaken, 2023-02-04:
Hi,
Next time please try to split topics into different issues, it makes it easier to manage it and keep track of the progress.

1. Yeah we can add a note regarding that in the documentation
2. We've been having some issues with the current dataset timestamp system so we might need to rethink it (@Aang23) 
3. That is planned for the future, see #67 
4. Yes we do plan on adding a frequency scale to the FFT. The second one would require some UI changes and creating an input field like that, which I am not sure whether it is in high enough demand to code it (takes up a lot of space, etc)

Aang23, 2023-06-28:
@sergsero is the UTC timestamp issue fixed currently? Or is it still of actuality?

sergsero, 2023-07-01:
Unfortunately, yes, the timestamp is wrong for some satellites.

My time zone is UTC+7. Sample datetime output format in my system (sequentially, at the time of writing this text):
```
~$ date
Sat  1 Jul 12:01:51 +07 2023
~$ date +%s%N
1688187809673653493
 ~$ echo $EPOCHREALTIME
1688187829.152689
```
Version SatDump: commit 57cd337.
Ubuntu 22.04 (5.19.0-46-generic #47~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Jun 21 15:35:31 UTC 2 x86_64 x86_64 x86_64 GNU/Linux)

For example, today (2023-07-01) NOAA-19, the estimated time when the satellite comes within the visibility zone of the ground station (according to local time UTC+7): 2023-07-01 08:58:19. And has left the zone in 2023-07-01 09:12:29.

In the directory named 2023-07-01_01-59_noaa_hrpt_1698Mhz (here the time was determined by the program correctly in UTC, not local, signal reception started at 08:59:01) I found the dataset.json file.

The time specified in it is the "satellite" timestamp: "NOAA-19", "timestamp": 1688248741.7626967.

```
~/data/2023-07-01_01-59_noaa_hrpt_1698Mhz$ cat dataset.json
{
    "products": [
        "AVHRR",
        "HIRS",
        "SEM",
        "MHS",
        "AMSU"
    ],
    "satellite": "NOAA-19",
    "timestamp": 1688248741.7626967
```

This is an erroneous time and indicates tomorrow:
```
$ date --date @1688248741.7626967
Sun 2 Jul 04:59:01 +07 2023
```
Although the error is only in hours, the minutes and seconds are indicated correctly.

It seems to me that the correct timestamp should be (excluded fractions of seconds) 1688176699 when NOAA-19 should be in sight soon. And 1688177549 when the satellite is about to pass out of range.

However, for Meteor N2-3 there the stamp (UTC time) was created almost correctly.

[image]

sergsero, 2023-07-02:
There were 5 sessions of morning data reception today, 02.07.2023 from 09:45 to 13:05 local time (UTC+7).
The Meteor N2-3 had a relatively correct timestamp.
FengYun-3C turned out to be in the 1970s. NOAA-19 was a few days and hours in the future.

[image]

Aang23, 2023-07-02:
Can you send over all those files?

sergsero, 2023-07-02:
> Can you send over all those files?

Should I send the dataset.json files? Or some others?

Aang23, 2023-07-02:
CADU / .raw16. That way I can process them locally and check what's going on with the timestamps.

sergsero, 2023-07-02:
https://drive.google.com/drive/folders/1TTFRfB2stik1FggCl1PM9xJRExM8RpHB?usp=drive_link

JVital2013, 2023-10-20:
When you get the chance, could you try updating to the newest version and try again?

We pushed a fix that should fix the timestamp bug in the viewer. Datasets created before updating will still be incorrect, but reprocessing from frames/cadus, and new captures, should be correct now.

The timestamp shown should be somewhere in the middle of the pass.

JVital2013, 2023-10-25:
Thanks for the info! Sounds like NOAA and MetOp are working correctly - and so does Meteor; see below. FY-3C seems to be the oddball. We'll do some digging on it - I can reproduce the issue with some sample data.

**For Meteor MRPT:** Unfortunately, this is expected behavior. The date is not encoded in the Meteor HRPT downlink - just the time. As such, the rough timestamp of the recording needs to be set in the UI on decode, as shown below (can be set via CLI as well). The exact time does not need to be correct - it just needs to be in the same day as the recording, and then SatDump will line everything up. NOAA and MetOp HRPT (and M2-3 LRPT) do not have this limitation, and the date will be understood correctly regardless.

[image]

JVital2013, 2023-10-25:
@meteosatellites @sergsero can you re-process FY-3C with the latest nightly? I found the bug so it should be fixed now.

[image]

meteosatellites, 2023-10-25:
Hello!

-- Отправлено из мобильного приложения Яндекс Почты

meteosatellites, 2023-10-25:
Hello! Have just downloaded the latest version of SatDump. Now the date in Viewer is correct when decoding FY-3C. Thank you very much! Good job!But I also mentioned that it failed getting TLE and can't make overlay. I use Windows version.Serge, Best regards!  25.10.2023, 21:58, "Jamie Vital" ***@***.***>***@***.*** @sergsero can you re-process FY-3C with the latest nightly? I found the bug so it should be fixed now.

JVital2013, 2023-10-25:
Glad to hear 3C's timestamp is working for you now!

About the TLEs, do you get an error when you go to settings and tell it to update? If so, can you share the log file? It would also be useful to know what you get when you go to http://celestrak.org/NORAD/elements/active.txt in a web browser - do you get TLEs, or an error?

meteosatellites, 2023-10-25:
I restarted SatDump and now it's ok. TLE uploaded. May be there were some issues wth server...And by the way could you please write where log file is situated. Can't find it via search ))

sergsero, 2023-10-26:
Beautiful job, thank you so much!
I usually receiving a data directly in Live mode (which was renamed to one of its functions - recording) and there I also found the new field for the timestamp when choosing the Meteor HRPT. Excellent.

And, if this issue will be closed, I wanted to ask about "The OpenCL with Intel's GPU" (ocl-icd-opencl-dev), from which it's title begins.

JVital2013, 2023-10-26:
@meteosatellites I'm glad to hear TLEs downloaded for you! Logs are at `%appdata%\satdump\satdump.log` on Windows for the installed version. For the portable version, they are in the same folder as the program.

@sergsero looks like we missed that detail - I added a bit to the install notes on installing the Intel OpenCL package. It's not a dependency of the SatDump deb since it won't work on systems without Intel iGPUs (or where the BIOS disables it, like mine does).

Your other requests in the original issue are being tracked in other issues (custom labels, better VFO support), so I will close this for now. Thanks again for all the feedback, and feel free to reach out again if something comes up!

closed 2023-10-26 by JVital2013