SatDump/SatDump #252: Frequency axis: Display of actual frequency instead of baseband frequency

eswnl, 2023-02-12:
**What would you want to be added to satdump?**
On the frequency axis, cursor displays baseband frequency instead of actual frequency. Is it worth changing the frequency axis to display actual RF frequency?

[image]

Aang23, 2023-02-20:
I agree. It's something we've been thinking about more and more. The main issue is, not too sure how to go about it UI-wise, to now have it interfere with the rest too much etc.

A VFO system will be coming for those narrower signals, in a similar style to other SDR apps.

ZbychuButItWasTaken, 2023-02-23:
Well what we have now is a placeholder while the VFOs are implemented. No changes will be made to the UI until VFOs are functional the the UI for that is implemented (because it will require a solid rewrite anyway)

closed 2023-07-02 by Aang23