SatDump/SatDump #255: Soft data with FY-3A/B/C DTP in offline processing.

Constellion, 2023-02-18:
I have got soft data after QPSK-modulator from FY-3B satellite and tried it with Release 1.0.3. 
But I don't understand what happen with my data or Release 1.0.3. Result of CADU data is none.

Aang23, 2023-02-18:
Do you know what format this soft data is in? If it's from a hardware or another demodulator, it's possible it is not the same 8-bit signed format expected by SatDump.

Constellion, 2023-02-18:
That's right. Thanks. 
In the next time, I will try with Yunhai satellite and I don't know the Release 1.0.3 support it ?

Aang23, 2023-02-18:
Yes, Yunhai is supported on L-Band. However, it is only down to CADU frames, as the payload is encrypted.

Constellion, 2023-02-19:
Would you like to try with lapan a3/Indonesia satellite. 
I have cadu frames. It is interesting topic too. All step by ccsds standard, but I can't continue at the decompression step.

Aang23, 2023-02-20:
Yes @Constellion if you have some data from Lapan, I'll gladly try it out! Actually tried to see it over here in Europe, but doesn't seem to transmit often at all.

Constellion, 2023-02-20:
Are you interested with Compressed image? 
I had cadu frames. All step by ccsds standard, cadu -> DE-PSEUDO-RANDOM SEQUENCE -> De-Rs -> Split channel, but I can't continue at the decompression step.

Aang23, 2023-02-20:
Definitely! I'd take CADUs (eg, before RS / derand perhaps) if possible. I've decompressed quite a few standards so I can see what I can do!
Perhaps if you have a short sample / specific of the downlink, I could also add a demodulator in SatDump in case someone else is able to receive it at some point.

Constellion, 2023-02-20:
Oh. Thanks. I will send you late. Perhaps in tomorow morning. Could you give me you email, I can send it over there. 😀

Aang23, 2023-02-20:
Thanks! aang23@altillimity.com

If you are able to receive this one, I do know quite a few other satellites you could try out transmitting in the area. Very few people seem to have such capabilities over there.

Aang23, 2023-02-20:
Also, about DPT. FY-3C is active, but the DPT format may differ a bit from 3A/B.

Constellion, 2023-02-22:
You had received cadu frames?

Aang23, 2023-06-28:
I'll close this - as the topic has move to something else.