SatDump/SatDump #260: Segmentation fault occurs during image transmission on Raspberry Pi (fengyun svissr)

pluggerlockett, 2023-02-26:
Hi Alan,

I'm attempting to get fengyun svissr live cli processing working on a Raspberry Pi, which works fine on my Nuc running Debian 11. I'm using the same command on both hosts. It appears to run/work fine on the pi until image transmission occurs. This is with `SatDump v1.1.0-f2b5bbfa`. Satdump release v1.0.3 does not display this issue but also doesn't save resultant images during live processing. 
I attempted to restart the live processing workflow and the crash occurred again. Error messages below.

```
...
[02/26/23 - 08:59:20] (I) Progress nan%                                         
[02/26/23 - 08:59:20] (I) Progress nan%, SNR : 8.499199dB, Peak SNR: 9.882178dB 
[02/26/23 - 08:59:30] (I) Progress nan%, SNR : 0.000000dB, Peak SNR: 9.882178dB 
[02/26/23 - 08:59:30] (I) Progress nan%                                         
[02/26/23 - 08:59:40] (I) Progress nan%, SNR : 8.171148dB, Peak SNR: 9.882178dB 
[02/26/23 - 08:59:40] (I) Progress nan%                                         
[02/26/23 - 08:59:50] (I) Progress nan%, SNR : 8.442638dB, Peak SNR: 9.882178dB 
[02/26/23 - 08:59:50] (I) Progress nan%                                         
[02/26/23 - 09:00:00] (I) Progress nan%, SNR : 8.209716dB, Peak SNR: 9.882178dB 
[02/26/23 - 09:00:00] (I) Progress nan%                                         
Segmentation fault                   

```
And after restart

```
...
[02/26/23 - 09:01:37] (I) Demodulating to /home/fy/fy-data/fengyun_svissr.soft
[02/26/23 - 09:01:37] (I) Buffer size : 163840
[02/26/23 - 09:01:37] (I) Start processing...
[02/26/23 - 09:01:37] (I) Using input symbols 
[02/26/23 - 09:01:37] (I) Decoding to /home/fy/fy-data/fengyun_svissr.svissr
[02/26/23 - 09:01:37] (I) Start processing...
[02/26/23 - 09:01:37] (I) Using input frames 
[02/26/23 - 09:01:37] (I) Decoding to /home/fy/fy-data/IMAGE
[02/26/23 - 09:01:40] (I) Progress nan%, SNR : 7.822359dB, Peak SNR: 8.607482dB
[02/26/23 - 09:01:40] (I) Progress nan%
Segmentation fault
```

[image]

Please let me know if I can do any testing to help with this issue.

Cheers,
drew

ZbychuButItWasTaken, 2023-02-27:
Thanks for reporting the issue.

Does it happen in offline processing too, if you use a frame file as input?

pluggerlockett, 2023-03-01:
I've not tried that, but I do have some .svissr files which I can test that with. But as an FYI, I ran into this issue again overnight. But the port and build were different. Last night at I was running the live cli functionality on a Debian 11 x64 build (running on an Intel NUC). I was using SatDump v1.0.3 Release. At 3pm UTC when image transmission started the live processing pipeline reported a Segmentation fault and then crashed. It had run fine for hours before without issue.

I've reverted back to using satdump-ui as I've never had it spit a segfault and I've been running it for some time without the same issue.

Aang23, 2023-03-09:
@pluggerlockett can you try it again with the latest? What PI is this running on?

pluggerlockett, 2023-03-30:
Hey @Aang23, sorry for the delayed response. Things have been a bit busy.
I reinstalled off master a couple days ago though and hopefully will set it up again tonight to test. Pi is a model 3B v1.2.

Aang23, 2023-04-20:
I'll close this for now, please comment if the issue persists.

closed 2023-04-20 by Aang23

pluggerlockett, 2023-04-20:
Oh man, I'm really sorry Alan. I did have another look and you 100% fixed the Seg Fault that was occurring during image transmission. But (and I'm not 100% on this so please don't hold me to it) I think there's a memory leak now. I tried multiple times to get the live cli processing to work, it seems to be working from the console output, but silently crashes and just stops providing console output before transmission ends.