SatDump/SatDump #262: Decode live TS.

eswnl, 2023-03-16:
**What would you want to be added to satdump?**
Option to decode live TS streams.
**Describe the feature you'd like to have**
If person wants to use separate DVB-S2 demodulator (software or hardware) and wants to output the live TS over UDP. 

**Describe alternatives you've considered**
At the moment, I have to record the TS stream first, then input it into SatDump manually.

**Additional context**
[image]
Possibly here?

Aang23, 2023-04-20:
Good idea indeed. I can't guarantee in the GUI though, the way "Recorder" works is very much for baseband inputs. However such a system can be added to work in CLI mode easily.

TennTrips, 2023-06-30:
How about input from Virtual Cable?
I really like the tracker and auto-scheduler in SDR Console and would like to port the audio from it into SatDump for decoding instead of WxtoImg which is very outdated..

hasaranga, 2023-12-23:
Yes, input from audio device is very useful.