SatDump/SatDump #265: Extra lines NOAA GAC Border Overlay

JVital2013, 2023-03-23:
**Description of the issue**
When adding borders to a NOAA GAC image with the "Viewer" feature, most borders are correctly placed, but there are always "extra" borders that show up all across the image. It happens with every GAC transmission I tested (NOAA-18 and NOAA-19).

I'm in the US so these are "backwards GACs," if that makes a difference.

**Hardware (SDR/PC/OS)**
Generic Msi2500/Msi001 SDR, running on multiple PCs. Tested Windows 10 and Ubuntu 22.04

**Version (Eg, 1.0.0, CI Build 171)**
f753584 (and a few older CI builds/self-builds I tried)

**Logs after the crash (satdump.logs)**
Nothing interesting in the log - here's the output when adding a border overlay:

```
[14:49:09 - 23/02/2023] (I) Update image...
[14:49:10 - 23/02/2023] (D) Filtering timestamps...
[14:49:10 - 23/02/2023] (I) Solving TPS equations...
[14:49:10 - 23/02/2023] (I) Forward solved
[14:49:10 - 23/02/2023] (I) Drawing map overlay...
[14:49:13 - 23/02/2023] (I) Done
```

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

**[FRM download](https://www.dropbox.com/s/xsiip03ua7mo3kf/2023-03-23%200132%20UTC%20-%20NOAA%2019.frm?dl=1)**

Example natural color image with bad borders:
[image]

ZbychuButItWasTaken, 2023-03-28:
Yep, this is a known bug that is a result of how we currently do projections. We are working on a fix (by basically rewriting the whole system)

Aang23, 2023-04-20:
Indeed a known issue. There are some patches but it's also simply a problem when using TPS transforms in this manner. A re-write of the geo-ref algorithm in the works for the edge cases.